frangoteam/FUXA #1116: OPCua write bool from script

robbudge, 2024-03-14:
with a OPCua tag of type bool and a script utilizing $setTag, a set value of true or 1 produces TypeError: value.toLowerCase is not a function in the log.

> TypeError: value.toLowerCase is not a function                                                                         │
│                                                     ││ main >     at _toValue (/home/FuxaPro/server/runtime/devices/opcua/index.js:714:27)                                           │
│                                                     ││ main >     at OpcUAclient.setValue (/home/FuxaPro/server/runtime/devices/opcua/index.js:395:31)                               │
│                                                     ││ main >     at Device.setValue (/home/FuxaPro/server/runtime/devices/device.js:241:14)                                         │
│                                                     ││ main >     at setTagValue (/home/FuxaPro/server/runtime/devices/index.js:269:52)                                              │
│                                                     ││ main >     at Object.IndexReset (/home/FuxaPro/server/runtime/scripts/msm-scripts.js:164:136)                                 │
│                                                     ││ main >     at MyScriptsModule.runTestScript (/home/FuxaPro/server/runtime/scripts/msm.js:52:47)                               │
│                                                     ││ main >     at /home/FuxaPro/server/runtime/scripts/index.js:75:43                                                             │
│                                                     ││ main >     at new Promise (<anonymous>)                                                                                       │
│                                                     ││ main >     at ScriptsManager.runScript (/home/FuxaPro/server/runtime/scripts/index.js:71:16)                                  │
│                                                     ││ main >     at /home/FuxaPro/server/api/scripts/index.js:42:36                                                                 │
│                                                     ││ main >     at Layer.handle [as handle_request] (/home/FuxaPro/server/node_modules/express/lib/router/layer.js:95:5)           │
│                                                     ││ main >     at next (/home/FuxaPro/server/node_modules/express/lib/router/route.js:144:13)                                     │
│                                                     ││ main >     at /home/FuxaPro/server/api/jwt-helper.js:66:17                                                                    │
│                                                     ││ main >     at /home/FuxaPro/server/node_modules/jsonwebtoken/verify.js:261:12                                                 │
│                                                     ││ main >     at getSecret (/home/FuxaPro/server/node_modules/jsonwebtoken/verify.js:97:14)                                      │
│                                                     ││ main >     at module.exports [as verify] (/home/FuxaPro/server/node_modules/jsonwebtoken/verify.js:101:10)

The write bool does fail.

unocelli, 2024-03-26:
Hi, Thanks for the report, we fix it as soon as possible.

unocelli, 2024-05-22:
This issue should be fixed in the latest version. let me know if you have any issues.