frangoteam/FUXA #1119: Can Timescale be supported

zk12668, 2024-03-19:
**Describe the feature**
Can Timescale be supported
Timescale Is very popular
Many projects are based on Timescale
Supporting Timescale allows for more projects to migrate over
**Describe the solution you'd like**
Describe your proposed solution here.

**Additional context**
If applicable, add screenshots to help explain your idea.

zk12668, 2024-03-19:
[image]

zk12668, 2024-03-20:
Timescale The database not only supports the temporal database,
The relational database is also supported,
It is much easier and more convenient to use,
Some of our previous projects were the Timescale,
I think some of the features are implemented with fuxa,
It would be nice to support the Timescale,
thanks very much!

unocelli, 2024-03-26:
Hi, fuxa support influxdb timeserie database for DAQ, for project and configuration a single file database like sqlite makes fuxa much more portable and light.

unocelli, 2024-05-22:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-05-22 by unocelli