frangoteam/FUXA #1127: Get two parameters for a function from script system function

IonuDragomir1, 2024-03-25:
Hi,
I have created and added a function on script system function. My function requires two parameters (two tag ids). So when I call the function on script editor, tag selection window is opening. Here I must select the parameters (the two tags), but I'm allowed to select only one. The second parameter I have to type manually or to call the $getTag function. What could I do to get both tags when calling my function?

unocelli, 2024-03-27:
Hi, sorry but I don't understand, can you send me an image that clarifies?

IonuDragomir1, 2024-03-28:
Hi,
My function requires  2 word tags as parameters.  When I click on `$WordsToReal`, tag selection window is opening.
In the below image, I can select only one tag. On tag selection window, I need to select both tags (WORD1 and WORD2).

[image]

What should I do to be able to select both tags from tag list?

IonuDragomir1, 2024-03-29:
Thanks for your reply!
I have modified on `script-editor.component.ts` multiSelection: true. Now I'm able to select both tags. 

[image]

My question is what should I do to get both of them called inside my function? This what I get after selecting both tags:

[image]

What I should get is: 
[image]

I have tried to modify on `client\src\app\_models\script.ts` the  `$WordsToReal`'s params in so many manners, but I couldn't get both tags. 

[image]

What should I do?

IonuDragomir1, 2024-03-29:
Thanks for reply!
I have modified on `scipt-editor.component.ts` multiSelection: true. Now I'm able to select both tags. But after selecting them, what I get is:
[image]

What should I get is:
[image]

I have modified on `client\src\app\_models\script.ts` $WordsToReal's params in so many manners but I couldn't succeed.
[image]

What should I do to get both parameters called on $WordsToReal function?

unocelli, 2024-04-13:
The issue is not so simple, because I would also like to be able to define in which order they should be bound

unocelli, 2024-10-27:
I’m closing this issue since there is nothing feasible

closed 2024-10-27 by unocelli