frangoteam/FUXA #1128: How to config the DAQ storage while using influxdb(2.0)?

jayjayhust, 2024-03-26:
I installed influxdb 2.0 and locally ran it on my PC([http://localhost:8086/]).
[image]

BUT in FUXA, how can i config the settings under the DAQ storage setting tab?
[image]

I didn't see any clues in the tutorial, maybe we can add up some instructions about DAQ storage in the tutorial in future.

Many thanks~

unocelli, 2024-03-27:
Hi, is working? 
We will make a video in fuxa channel [youtube]( https://www.youtube.com/@umbertonocelli5301)

jayjayhust, 2024-04-05:
> Hi, is working? We will make a video in fuxa channel [youtube](https://www.youtube.com/@umbertonocelli5301)

Great~ 
It's still not working. Looking forward to your update on youtube :)

unocelli, 2024-04-16:
https://youtu.be/UuE_uLSXWGM?si=szvDja1O-OC5ynNR

jayjayhust, 2024-04-17:
> https://youtu.be/UuE_uLSXWGM?si=szvDja1O-OC5ynNR

Nice to hear the latest update, cheers~

closed 2024-04-17 by jayjayhust