frangoteam/FUXA #1158: Templating

bogdanvaduva, 2024-04-19:
I will start to say what a great project it is!
**Describe the feature**
Being a SCADA application it would be great to have some way of building templates. Let's say we have one item (IoT or something similar) that is replicated for 200 times (as an example). At this point the only way of doing it it is to create 200 identical Views which is not great. 

**Describe the solution you'd like**
The way I did it was to branch your project and change a few things.
1. I allowed page query parameters to become View variables by default, that means that any query parameter will become a variable with the same name
2. I created a generic a convention called tagName which can be use throughout the Views as variable/tag
3. The tagName convention is replace at runtime with the query param value of tagName which means that the View is getting an actual tag at runtime
4. I created a repeater for allowing me to have 200 inputs if I want.

**Additional context**

[image]

[image]

[image]

[image]

MatthewReed303, 2024-04-19:
@bogdanvaduva did you see https://github.com/frangoteam/FUXA/wiki/HowTo-use-same-view you can use the same view as a pop up etc

bogdanvaduva, 2024-04-20:
Yes @MatthewReed303,  I did see it (Thanks). The multiview container required me to define the 200 views and  assign each view an actual tag which would be quite a task. I wanted something easier like a data source (for the 200 tags) and a way of assigning that data source to a View. Thats the way I did it and I don't say it's perfect, I just wanted to share.

Jue07, 2024-04-22:
Hi, @bogdanvaduva. Thanks for this interesting sharing. Can you share the details of the steps in a video on how this feature works? I'm still not clear on this. Thank you so much in advance.

bogdanvaduva, 2024-04-22:
Hi @Jue07 , I've made a short video. Hope it helps!

[image]

Jue07, 2024-04-22:
Thank you so much @bogdanvaduva. May I know which version of Fuxa you are using? because I am using Fuxa 1.1.18 there is no Repeater Data Settings feature.

MatthewReed303, 2024-04-22:
@bogdanvaduva why not create a custom Fuxa Branch for your features and then create a pull request and then @unocelli can look at them and merge them into the master branch?

closed 2025-10-03 by unocelli