frangoteam/FUXA #1174: [FEATURE] Proposal for Dynamic Shape/Resource Element to Simplify State-based Image Handling

mauroalexandre, 2024-05-02:
#### Summary
In various projects and scenarios, there is a recurring need to display different images that correspond to distinct values for the same tag. To simplify this process, I propose the creation of a dynamic shape/resource element.

#### Description
With this new element, users would be able to define which resource should be displayed for each state/value of the tag. Currently, our approach involves overlaying multiple images and toggling their visibility depending on the tag's state/value. This method is not only impractical but also makes maintenance and organization challenging.

#### Current Challenges
- Multiple overlaid images require manual visibility adjustments.
- Organizational and maintenance efforts are significantly increased.

#### Proposed Solution
Implement a dynamic resource element where:
- Each state/value of a tag corresponds to a different image or resource.
- Simplifies the UI by reducing the number of elements needed on the screen at any one time.
- Enhances maintainability and organization of resources.

#### Contribution
While I lack the technical expertise to implement this feature alone, I am eager to contribute in any way necessary to help develop this functionality.

#### Additional Materials
Attached is a suggestion on how the property screen of this element might look, illustrating the proposed behavior.

Thank you for considering this enhancement. I look forward to the community's feedback and hopefully, collaboration on this feature.

[image]

MatthewReed303, 2024-05-03:
@mauroalexandre The way you suggest is still overlaying the images? I do this currently by show/hide based of values and then group them together, to reuse I just copy that group and do an edit bind of tag to change the tag for the whole group. 

If we could save the group as a symbol and be able to export/import it as a json symbol would be great and save time. 

To be fully dynamic I think the best approach would be to have symbol builder or a way to define elements of the SVG and have a placeholder for a tag and then assign actions to that element of the SVG, change colour/rotate/show/hide/blink/move etc most of this Fuxa already has under actions except for change colour.

A quick workaround could be you edit your SVG in a text editor or Inkscape and add the place holder to the elements you want to alter, Fuxa then looks for these placeholders in the SVG file and allows you to bind a Tag to the placeholder. I spoke to @unocelli about this and the past and he said the current SVG editor can not use the SVG elements and it's a big job to update the SVG editor it's self, so maybe a placeholder in the file could work? What do you think @unocelli ?

mauroalexandre, 2024-05-06:
@MatthewReed303  I am not very familiar with the structure of the Fuxa project storage, but ideally, it would be a collection of resources, within this "new element," where each resource would be a key-value pair.

Something similar to:

```json
{
  "name": "dynamic_shape_1",
  "tag": "tag_db_1",
  "states": {
    "0": "<svg data from image 1>",
    "1": "<svg data from image 2>"
  }
}