frangoteam/FUXA #1177: [FEATURE] Cron and Condition functionality (if...then)

maufc, 2024-05-04:
**Describe the feature**
I would suggest to add a Cron system and a condition functionality (if...then) in order to schedule task for the devices/tags.

**Describe the solution you'd like**
Program tasks and take actions on specific events or critical values.

Thanks!

MatthewReed303, 2024-05-04:
@maufc you can currently schedule your scripts. But I agree we need a proper scheduler that can be used from the UI by the end user. This has been discussed a few times. 

[image]

I was using Node-Red scheduler in a iframe at one stage but was a few limitations with the iframe ( any remote viewing of Fuxa on different network etc needed an connection to Node-Red )

unocelli, 2024-05-17:
Hi, about condition the I suppose that the feature will be available when we have implemented to set a 'deadband' on the tags for DAQ registration, then we can add to trigger a script too.