frangoteam/FUXA #1185: Feature - Alarm Action - Run Script

unocelli, 2024-05-09:
### Discussed in https://github.com/frangoteam/FUXA/discussions/1184

<div type='discussions-op-text'>

<sup>Originally posted by **martinjug** May  9, 2024</sup>
Hi,

This feature would enhance flexibility when alarms occur, allowing for various actions to be triggered. For instance, it could enable sending SMS messages via an SMS API or calling other HTTP APIs. Additionally, it could facilitate the execution of more complex procedures in response to specific events.

Regards,
Martin

[image]

</div>

mauroalexandre, 2024-11-07:
@unocelli I checked here, and this option already seems to have been implemented. This issue should be closed, right?

unocelli, 2024-11-10:
@mauroalexandre Yes, Thank you, I normally close them when I do the release, for the time being I change the label so you can see what has been implemented (solved or fixed)

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli