frangoteam/FUXA #1190: [BUG] Toggle value action not working always

IonuDragomir1, 2024-05-14:
Hi,

Please see the below video:

https://github.com/frangoteam/FUXA/assets/58516894/8680a158-de4e-4de6-b44a-7b90faeef8ba

I am running FUXA v1.1.18 on Windows 10 PRO using Chrome browser. The error might appear also when using external tags.

unocelli, 2024-05-18:
Hi, you have to set the init value, without init the value is null, not true and not false

unocelli, 2024-05-22:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-05-22 by unocelli

maxopa, 2024-12-06:
Hi, i have the same issue(
TAGs receive data via ModBusTCP. TOGGLE does not work until i set it to 1 manualy, but the initial value of the tag is 0. Also helps to change the modbus address and change it back.