frangoteam/FUXA #1191: [BUG]Failed to Write the Real or Dword Type Register address of the S7 Device

HugeShao, 2024-05-14:
**Describe the bug**
Failed to Write the Real or Dword Type Register address of the S7 Device
If I Use Int or Word Type Register address  Write to the S7 Device can succeed！
[image]
[image]

**Environment**
 - FUXA version: 1.1.19-1592
 - Node.js version: v14.21.3
 - npm version:6.14.18
 - Platform/OS: Windows 10
 - Browser: Google Chrome

unocelli, 2024-05-18:
Hi, I can't confirm the issue, I tested with my simulator and are working for both type

unocelli, 2024-08-23:
I assume the issue has been resolved in the meanwhile. I'm therefore going to close this issue.

closed 2024-08-23 by unocelli