frangoteam/FUXA #1192: [BUG] Intergation in docker-compose shows error in manifest

MikeEitel, 2024-05-14:
**Describe the bug**
I can not get docker-compose to run fuxa

**To Reproduce**
Steps to reproduce the behavior:
docker-compose.yml 
......
  fuxa:
    container_name: fuxa
    image: ghcr.io/frangoteam/fuxa:latest
    restart: unless-stopped
    ports:
    - "1881:1881"
    volumes:
      - ./volumes/fuxa/appdata:/usr/src/app/FUXA/server/_appdata
      - ./volumes/fuxa/db:/usr/src/app/FUXA/server/_db
      - ./volumes/fuxa/logs:/usr/src/app/FUXA/server/_logs
      - ./volumes/fuxa/images:/usr/src/app/FUXA/server/_images
    networks:
    - iotstack_nw

**Expected behavior**
following error message........
pi@PI5:~/IOTstack $ docker-compose up fuxa
[+] Running 1/1
 ✘ fuxa Error manifest unknown                                                                                0.6s 
Error response from daemon: manifest unknown

when using with docker alone and not in my compose environment I got it loaded and running

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version:  latest
 - Node.js version:  latest
 - npm version: latest
 - Platform/OS: lastest RPIOS Pi5
 - Browser: edge/chrome

MikeEitel, 2024-05-15:
I solved it
image: **ghcr.io/**frangoteam/fuxa:latest
I had to get my personal token activated

MikeEitel, 2024-05-15:
I had to get my personal token activated

unocelli, 2024-05-22:
I’m going to close this as resolved. let me know if you have any issues.