frangoteam/FUXA #1193: Multiple Slave on single IP Modbus TCP can connect only one slave

Safe2024, 2024-05-15:
When a Modbus Gateway is used for connecting multiple slaves on single IP only one slave data is recieved.

I used a mgate as a gateway to connect multiple energy meters on single IP 
I created multiple modbusTCP devices and connected all it showed connected with different slave IDs.
But when I click on link and add register no data is recieved.
I also tried fragmented and also buffered port but faced same issue.

As it must scan every slave using I'd one after the another .

[image]

**Environment**
 - FUXA version: 1.1.18-1445
 - Node.js version: 16.20.2
 - npm version:8.19.4

Safe2024, 2024-05-15:
[image]

rvbatista, 2024-05-15:
Same as #211

Safe2024, 2024-05-16:
I have test some things found that for multiple IDs it creats different connections even if it has same IP.
The grouping must be done under same IP in which scaning can be better.

unocelli, 2024-08-23:
I suppose that is the feature added from @rikugun (#1312) is now available in master branch

unocelli, 2024-12-11:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-12-11 by unocelli