frangoteam/FUXA #1196: [BUG] - User login not showing?

killerbeat12, 2024-05-17:
**Describe the bug**
i added the following to  'server\_appdata\settings.js'
secureEnabled: true,            // enable or diasable
secretCode: 'frangoteam751',    // secret code to encode the token
tokenExpiresIn: '1h'            // token expiration delay '1h'=1hour, 60=60seconds, '1d'=1day

But i don't get the user login also not with restarting, what do i do wrong?

**Expected behavior**
a login screen will appear when going to the fuxa ip adres + port.

**Environment**
 - FUXA version: 1.1.18
 - Node.js version: 14.21.3
 - npm version: 6.14.8
 - Platform/OS: Windows Server 2022
 - Browser: FireFox/Chrome/Brave

mauroalexandre, 2024-05-17:
are you using docker?

-- 

*Mauro Alexandre*
*Analista de Tecnologia*
*Telefone: (34) 9 8420-7007*
*www.webrota.com.br <http://www.webrota.com.br/>*

O conteúdo deste e-mail ou seus anexos são elaborados exclusivamente ao
destinatário e contém informações confidencias e/ou legalmente protegidas.
Caso não seja o destinatário, é proibido sem autorização, ler, revelar,
distribuir, alterar, copiar, reproduzir ou, sob qualquer forma, utilizar o
todo ou parte desta mensagem ou dos documentos a ela anexados. Não nos
responsabilizamos por opiniões e/ou declarações veiculadas por e-mail não
ficando obrigada ao cumprimento de qualquer condição constante deste
instrumento.

unocelli, 2024-05-18:
Hi, enable show login on start and set overlay color to disable cancel button in Login form
[image]

killerbeat12, 2024-05-18:
[image]
Thanks for your reply, i already had  show login on start enabled but still not working, i don't have the overlay option with my version?

killerbeat12, 2024-05-18:
no i am not using docker, i just using the download file from the github and installed and start it with npm...

unocelli, 2024-05-18:
sorry, you are right...is coming in the next version

killerbeat12, 2024-05-20:
Do you have any idea when next version is coming out?

unocelli, 2024-05-22:
is now available in the latest version, let me know if you have any issues.

closed 2024-05-22 by unocelli

Vau9hn71, 2024-06-07:
Hi,

I am having the same problem. Done as instructed, but login screen never shows - using Version 1.1.19-1619

Is there anything else that needs to be done to enable security?

Thanks,
Vaughn

killerbeat12, 2024-06-07:
i got it working in the latest version, also the one which you are using, steps:
1. add this to the setting file 'server_appdata\settings.js'
  secureEnabled: true, // enable or diasable
  secretCode: 'frangoteam751', // secret code to encode the token
  tokenExpiresIn: '1h' // token expiration delay '1h'=1hour, 60=60seconds, '1d'=1day
2. see picture from unocelli, enable show login on start and set overlay color to black.

Vau9hn71, 2024-06-07:
Thank you for the quick reply, It works perfectly now.

wonka929, 2024-09-13:
No login again, docker version