frangoteam/FUXA #1200: [Question] Script scheduler

AllmarkPL, 2024-05-21:
Hello,
quick question about script scheduler - how many times should the script be executed by server when set on specific hour? I assumed that if I set it e.g. for 6:00 it will be run just once, but it appears that it continues to run for whole hour and then stops... I set a simple script which get current value from device, then stores it into tag, and then get value from previous hour and substracts one from another (basic delta value), but as the script keeps on going the value keeps on incrementing...

I'm attaching script code and graph for this issue. 

[image]

[image]

[image]

unocelli, 2024-06-02:
Hi, Thanks for the report

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli