frangoteam/FUXA #1201: [BUG] - Import/Export CSV MQTT doing something wrong with the actual value and timestamp?

killerbeat12, 2024-05-25:
**Describe the bug**
i want to import and export tags, i have like 10 the same devices on MQTT and just want to copy the same tags with only different tagname, mqtt topic and UUID.

For some reason i don't get this to work and i just made a simple video where i export the working tags, copy 1, change name, topic and uud and then import. 

After that all tags have strange value information.

**To Reproduce**
Steps to reproduce the behavior:
1. See video

**Expected behavior**
just 1 new tag.

**Screenshots**
see attachment

**Environment**

https://github.com/frangoteam/FUXA/assets/60966783/4371c311-fe6d-4563-8438-0c69a1f61bca

 - FUXA version: 1.1.19
 - Platform/OS: Server 2022
 - Browser: Chrome/Firefox

unocelli, 2024-06-02:
Hi, Thanks for the report

zhanjiayi, 2025-07-03:
So how was this problem ultimately solved?