frangoteam/FUXA #1203: [BUG] Bacnet - some Tags don't show up

TheGreyDiamond, 2024-05-25:
**Describe the bug**
Some tags do not show up when using bacnet. I've integrated a Siemens POL908 Climatix controller (for an HVAC system). However, not all fields show up. Notably the `OpModeAutoManStTmpSwtch` option (Used to set operating mode) wont appear. Using the Python BAC0 library I can discover and control this endpoint. It shows up as a "Multiselect" (or similar - at the time of writing the system has gone offline for an unknown reason and I can only remote in) type, which I assume to be part of the problem.

**Environment**
 - FUXA version: Version 1.1.19-1500 (from bacnet-Branch)
 - Node.js version: 18.20.3
 - npm version: 10.7.0
 - Platform/OS: Rapsberry Pi 3 / Debian

unocelli, 2024-06-02:
Hi, Thanks for the report.
@lpaszcza do you have any idea what could be the problem?

lpaszcza, 2024-06-03:
@TheGreyDiamond I suspect that you are trying to read an MV type object that is not supported in the current version.
@unocelli I've added a new PR to the bacnet branch, which includes a fix that adds support for MultiVariable points and others.

TheGreyDiamond, 2024-06-03:
Sounds, good.  Currently downloading and installing your PR, will hopefully be able to try it later today..

TheGreyDiamond, 2024-06-03:
Trying your branch I'm now unable to find any endpoints.. I can discover the node and this screen shows up:
[image]

Trying to open the last dropdown I get the following error in the log:
```
[ERR]  device-browse: undefined
```
This is my device config:
[image]

lpaszcza, 2024-06-03:
@TheGreyDiamond, thanks for your feedback. I need to investigate this case further. Could you tell me how many BACnet objects you are trying to read from the Siemens device?

TheGreyDiamond, 2024-06-04:
Sorry for the late reponse. BAC0 reports 94 points.

TheGreyDiamond, 2024-06-11:
Hey there. Fairly certain that there are AV and BV-type objects. The BACnet branch seems them. The master branch of FUXA did not work for me (at all, I couldn't get Bacnet to turn green).

I've tried multiple setups and reseted it a few times.