frangoteam/FUXA #1204: How to use ODBC on Fuxa?

IonuDragomir1, 2024-05-28:
Hi,
I have seen that in the latest version was added ODBC plugin. 
I have created a new ODBC Data Source.

[image]

How can I use it in FUXA?

MatthewReed303, 2024-05-29:
@IonuDragomir1 it's not in the latest release yet, will be in the next. Check out the ODBC branch. Are you using docker? I recommend doing a docker build from the odbc branch. The docker build will setup and install all the odbc drivers for all the major databases such as mySQL, PostgreSQL, MSSQL etc in the docker file you will need to add on the git the branch, -b odbc to pull the odbc branch rather than the master before you do the build. 

Download the docker file from:
https://github.com/frangoteam/FUXA/blob/odbc/Dockerfile

Edit in docker file to:
```
# Clone FUXA repository
RUN git clone -b odbc https://github.com/frangoteam/FUXA.git
```

Build the docker file:
`sudo docker build -t fuxa-odbc .
`
You will now have an image called fuxa-odbc you can now use in your docker compose fine.

```
version: '3.5'

services:
    fuxa:
        image: fuxa-odbc
        network_mode: "host"
        volumes:
            - ./fuxa_appdata:/usr/src/app/FUXA/server/_appdata
            - ./fuxa_db:/usr/src/app/FUXA/server/_db
            - ./fuxa_logs:/usr/src/app/FUXA/server/_logs
        environment:
            - TZ=Pacific/Auckland
        restart: always
```

You can now start it using `sudo docker compose up -d` note you can see in above example I use host network mode so you have direct access to devices on your network likes PLCs and Database etc

You can then setup a odbc device and a connection string for you DB
`DRIVER=PostgreSQL;SERVER=localhost;PORT=5432;UID=admin;PWD=mypassword;DATABASE=testDB`

Now you can create a server side script and make a query: 
Hi Matthew, I updated the odbc branch, new there is a sys function getDevice to use in script:
```
let device = $getDevice('odbc', true);
let result = await device.pool.query(`SELECT * FROM SampleTable`);
```
You can then put the data into a table, chart etc write to the DB

Kapket, 2024-06-17:
Where do I put connection string? In odbc.ini?

MatthewReed303, 2024-06-17:
In the Fuxa Devices ODBC settings in the UI

Kapket, 2024-06-17:
[image]
Here, right?
But there is no place for port.

MatthewReed303, 2024-06-17:
@Kapket just put the complete string in the Data Source Name field. 

@unocelli can we add Fields for all of these parameters to make it easy. `DRIVER=PostgreSQL;SERVER=localhost;PORT=5432;UID=admin;PWD=mypassword;DATABASE=testDB`

abishur, 2024-06-21:
I was following the advice in this issue, but can't seem to get connected to a database.  I've tried with both mySQL and PostgreSQL.  With mySQL I was able to connect to it from another computer using mySQL workbench, but within FUXA it kept saying there was an issue connecting to the database.  In the DSN field I put

`DRIVER=MySQL;SERVER=localhost;PORT=3306;UID=myuser;PWD=mypassword;DATABASE=mySchema`

When I tried with Postgre I used the example Matthew lists above.

Also, I did have to change the Dockerfile.  Line 9 and Line 47 both have you trying to clone the FUXA repository, in your docker image which causes it to fail when you get to line 47.  I replaced line 9 with line 47 and commented out the existing line 47 and it worked.

Then when I created the container from the image I made sure to pass in port 1881 so I could still reach the web server.  Are there any other ports or volumes I needed to map in to make sure it can each my SQL server?

MatthewReed303, 2024-06-21:
@abishur we are currently fixing dockerfile now

If you execute into the container it's self can you ping your DB IP/host? `SERVER=localhost` this needs to be your database IP address or hostname if dns setup. 

If you database is running in docker and not host networking mode, you need to add that network to the Fuxa container so it has access, please try with Fuxa in host network mode and database in host network mode and use the IP address of the PC you have them running on.

abishur, 2024-06-21:
the mysql database is running on the host machine (not in a docker container).  I changed the ip address from localhost to the ip of my host machine and confirmed that I can ping it from the docker container.

I'm still getting an error message that says "getendpoints-error: Error: [odbc] Error connecting to the database

MatthewReed303, 2024-06-21:
@abishur here are the correct names for the drivers https://github.com/frangoteam/FUXA/blob/master/odbc/odbcinst.ini 

DATABASE=mySchema are you sure this is the database name?

You can ping the host, but firewall not blocking access to the port? `nmap -p 3306 yourIP/host` from the Fuxa container ( may have to install nmap in the container )

abishur, 2024-06-22:
I installed nmap and it reports 1881 and 3306 are both open (which makes sense since I can connect to the database from another PC using workbench)

I've double checked permissions and table names and everything looks like it should be connecting, but it keeps giving me that error connecting to the database message

MatthewReed303, 2024-06-22:
@abishur I have just done a test with PostgreSQL on latest master, please build the latest dockerfile with `sudo docker build -t fuxa-odbc --no-cache .`

[image]

Also you now need to enter the user and password in the fields and remove from DSN connection string

I will spin up mySQL and give that a go also

abishur, 2024-06-23:
I had some issues building the docker file on my Pi zero 2 w, but once I swapped over to a Pi 4, it built fine.

It still won't let me connect to MySQL (I removed the UID and PWD fields from the DSN string but kept the rest the same), but I'll try installing PostgreSQL this evening and see if it works then.

abishur, 2024-06-23:
I can confirm that after making the docker image on the Pi 4, I was able to transfer it to my Pi Zero 2 W and connect to the PostgreSQL server!  🎉  There appears to be some issue with MySQL at this point in time, but I'm counting this as a win.

PostgreSQL is a new one for me, so it took me a minute to realize I needed to edit the file
`/etc/postgresql/15/main/postgresql.conf`
file, uncomment the listen_address line and change the 'localhost' to '*'
Then I had to edit the file
`/etc/postgresql/15/main/pg_hba.conf`
file and add 
`host all all 0.0.0.0/0 md5`
I could make this more secure by entering the subnet information for my docker machines, but this is good for now.

MatthewReed303, 2024-06-23:
@abishur is your Pi Zero 2 W 32bit? mySQL ODBC driver does not support arm 32bit 
I'm just doing a build and installing mySQL now and will test and see if I can get it to work

I did not have to change any of that for postgresql, I used docker compose along with pgadmin

```
version: "3.8"

services:
  db:
    image: postgres
    container_name: local_pgdb
    restart: always
    network_mode: "host"
    environment:
      POSTGRES_USER: user
      POSTGRES_PASSWORD: mypass
    volumes:
      - local_pgdata:/var/lib/postgresql/data
  pgadmin:
    image: dpage/pgadmin4
    container_name: pgadmin4_container
    restart: always
    ports:
      - "8888:80"
    environment:
      PGADMIN_DEFAULT_EMAIL: user
      PGADMIN_DEFAULT_PASSWORD: mypass
    volumes:
      - pgadmin-data:/var/lib/pgadmin

volumes:
  local_pgdata:
  pgadmin-data:
```

abishur, 2024-06-23:
I'm using Pi OS Lite (64-bit) released 2024-03-15.  I had to make the  PostgreSQL changes since installed it directly on the host machine and not in a docker.  I *HAD* hoped to cut down on any lurking gotchas by not having it in a docker container, but I ended up creating a couple new ones for me to deal with by installing it on the host, lol.

Hiemer23, 2024-06-24:
`let device = $getDevice('odbc', true);`
`let result = await device.pool.query("SELECT * FROM SampleTable");`
I tested the following code, but it did not work correctly. The expected result was that the query would return the appropriate value, but it returned nothing and only displayed [object Object] "{}". What is the correct method to access the database information?
[image]
[image]

MatthewReed303, 2024-06-24:
@abishur The MySQL drivers were not being installed correctly, I have created a PR https://github.com/frangoteam/FUXA/pull/1246 

I have also added mariaDB ODBC, you will just need to wait for @unocelli to merge it. 

There is still an issue with Fuxa/node-odbc where it crashes using MySQL driver, the driver is working from within the Docker container. You can test yourself

`myodbc-installer -s -a -c2 -n "test" -t "DRIVER=MySQL;SERVER=YourIP;PORT=3306;DATABASE=testDB;UID=User;PWD=MyPass"`

Then

`isql test`

You can then test your SQL commands from the console

MatthewReed303, 2024-06-25:
Hi @abishur I have now got MySQL driver now working, the crash is caused by `libssl` if you add `;SSLMODE=DISABLED` to the end of the connection string `DRIVER=MySQL;SERVER=YourIP;PORT=3306;DATABASE=testDB;SSLMODE=DISABLED` 

You will have to just wait for the PR to be done by @unocelli or you can try it from my Github fork, just be sure to change the git clone on the docker file from the Fuca github to my github

hxjackcn, 2024-07-03:
> Hi @abishur I have now got MySQL driver now working, the crash is caused by if you add to the end of the connection string `libssl``;SSLMODE=DISABLED``DRIVER=MySQL;SERVER=YourIP;PORT=3306;DATABASE=testDB;SSLMODE=DISABLED`
> 
> You will have to just wait for the PR to be done by @unocelli or you can try it from my Github fork, just be sure to change the git clone on the docker file from the Fuca github to my github

DRIVER=MSSQL;SERVER=192.168.33.136;PORT=1433;DATABASE=etong_rules

The database is correct；I can connect to the DSN that comes with the local operating system, but I cannot connect to it in fuxa.

MatthewReed303, 2024-07-03:
@hxjackcn can you try the FreeTDS driver

`DRIVER=FreeTDS;SERVER=192.168.33.136;PORT=1433;DATABASE=etong_rules`

Please also go through all the checks in the above steps to debug. Please make sure you are running the latest master and have done a correct docker build, all the drivers were fixed. 

Check you can ping the MSSQL server from within the docker container and port is open with nmap

hxjackcn, 2024-07-04:
I use it under Windows, I don't need docker.
[image]

There is no problem with the remote database server port via telnet, and a connection can be established via other software.

hxjackcn, 2024-07-04:
The latest compiled fuxa is used.
[image]

MatthewReed303, 2024-07-04:
@hxjackcn you need to run it in docker to use the built in drivers, the docker build will install and setup the odbc drivers. See the install script here: https://github.com/frangoteam/FUXA/blob/master/odbc/install_odbc_drivers.sh and these are there drivers in Linux https://github.com/frangoteam/FUXA/blob/master/odbc/odbcinst.ini 

If you are using directly in windows you will have to use the DSN name of your connection, the Linux drivers will not work. @unocelli had it working within windows using DSN, maybe he can guide you here. I recommend using docker, you can install docker on windows or WSL on windows and run it there.

Kapket, 2024-08-22:
Hi, I have to say that the current version is actually better than the previous one, as the problems with installing the plugin have disappeared. However, using it itself, continues to be impossible for some reason. Connecting to the database works, but the error “Error getting table information” appears. We tried connecting on multiple databases (MySQL, SQLite, MariaDB, PostgreSQL) and on several systems (Ubuntu, Windows, Debian).

We have no idea where this error is coming from. Is it possible that ODBC has a problem with current database versions?

MatthewReed303, 2024-08-22:
@Kapket where are you seeing `Error getting table information` are you connecting and reading tables via SQL query in a script? What does console.log say when returning the query? I'm still yet to do the Wiki for how to use correctly. I will try to add some details tomorrow.

Kapket, 2024-08-22:
I see this error in result of table found
[image]
Console log displays that there is a connection to the database, which is actually true. When I try to display the connection in the field for numbers, it shows me a status of 5

MatthewReed303, 2024-08-22:
@Kapket please ignore that and try to create a script with a SQL Query and use consol.log to display the results. You can then push the results directly to a Table via the script. If you need some examples I will get them all together tomorrow.

Kapket, 2024-08-22:
Yes, examples would be great.

Kapket, 2024-08-23:
Is it possible to access device tags from odbc in connections? Just like you can do this with WebAPI device.

nex2000, 2024-08-26:
I'm trying to connect to a mysql database I used this connection string
DRIVER=MySQL;SERVER=YourIP;PORT=3306;DATABASE=testDB;SSLMODE=DISABLED

I always get the same error message

[image]

But as advised I'm ignoring it, after these steps how can I read a value? I created a query in a script but I'm sailing in the dark, a post would be useful that shows how to get data from a database, thanks.

NEX

MatthewReed303, 2024-08-26:
Hi,

First you need to get ODBC connection via devices, in my example it's named postgreSQL

```
// Initialize the device, sane name as connection
let myDevice = await $getDevice('postgreSQL', true);
```
Read data from the DB
```
let result = await myDevice.pool.query(`SELECT * FROM "testTable"`);
console.log(JSON.stringify(result));
```
Here is an example with Tags

```
await myDevice.pool.query(`INSERT INTO "testData"."Customer" ("Name", "Phone", "Email", "Age") VALUES ('${customerName}', '${customerPhone}', '${customerEmail}', ${customerAge})`);
```
If you need to poll tags and write to DB using a trigger see this full example, we use a one shot to execute the query:

```
// Query Manager, function to provide one shot based on trigger event
async function createQueryManager(device) {
    let lastTriggerState = false;
    return async function(trigger, sqlQuery) {
        if (trigger && trigger !== lastTriggerState) {
            try {
                const result = await device.pool.query(sqlQuery);
                lastTriggerState = trigger;
                return result;
            } catch (error) {
                return 'Error executing query';
            }
        }
        lastTriggerState = trigger;
    };
}

// Initialize the device, sane nane as connection
let myDevice = await $getDevice('postgreSQL', true);

// Create query Manager for each query type, retains query data
let executeInsertQuery = await createQueryManager(myDevice); // Instance Insert
let executeSelectQuery = await createQueryManager(myDevice); // Instance Select

// Global Variables to retain Data
let selResult;

// 100ms loop catch Tag Events
let myLoop100ms = setInterval(loop100ms, 100);

async function loop100ms() {
  
    let customerName     = $getTag($getTagId('customerName'));
    let customerPhone    = $getTag($getTagId('customerPhone'));
    let customerEmail    = $getTag($getTagId('customerEmail'));
    let customerAge   	 = $getTag($getTagId('customerAge'));
    let execSaveCustomer = $getTag($getTagId('execSaveCustomer'));
  
	// Call Query Manager Instance Function Every 100ms, first parameter is the trigger to execute the query once with a one shot
    await executeInsertQuery(
        execSaveCustomer,
        `INSERT INTO "testData"."Customer" ("Name", "Phone", "Email", "Age") VALUES ('${customerName}', '${customerPhone}', '${customerEmail}', ${customerAge})`
    );
  
    // Testing second instance calling same function type
    //selResult = await executeSelectQuery(
    //    execSaveCustomer,
    //    'SELECT * FROM "testData"."Customer"'
    //);
	//
    //$setTag($getTagId('customerDataArray'), JSON.stringify(selResult));
}

// 1 sec loop update data from DB
let myLoop1sec = setInterval(loop1sec, 1000);

async function loop1sec() {
  // No Need to use query manager here as we are reading from DB every 1 sec to update Data for the Table
  selResult = await myDevice.pool.query('SELECT * FROM "testData"."Customer"');
  $setTag($getTagId('customerDataArray'), JSON.stringify(selResult));
}
```
You can create a client side script with 1 sec interval and write to a table using the below:
```
let customerData = JSON.parse($getTag($getTagId('customerDataArray'  ,)));

// Name of table used in Fuxa, column ID's must match DB column
var tableData = {
  columns: [{
    id: 'Name',
    label: 'Name'
    }, {
    id: 'Phone',    
    label: 'Phone'
    }, {
    id: 'Email',    
    label: 'Email'
    }, {
    id: 'Age',    
    label: 'Age'
  }],
  rows: customerData
};

$invokeObject('customerTable', 'setTableAndData', tableData);
```

nex2000, 2024-08-26:
Thanks Matthew for your quick reply, but unfortunately I'm a beginner and I'm not able to do anything similar, I thought there was an example inside fuxa so I could replicate the code.

I'm using a linux operating system on a virtual machine I installed Fuxa and my database for the rest I have some knowledge of SQL but nothing more.

Greetings

MatthewReed303, 2024-08-26:
Hi @nex2000 

Did you install using Docker? 

If not you will need to run the install script located here if system is Debian based:

https://github.com/frangoteam/FUXA/tree/master/odbc

Make script executable:

`sudo chmod +x install_odbc_drivers.sh`

Install the script:

`sudo ./install_odbc_drivers.sh`

This will install the needed ODBC drivers for DBs defined in the ini

You also need to copy the ini file into etc

`cp odbcinst.ini /etc/odbcinst.ini`

nex2000, 2024-08-26:
I installed everything on Ubuntu/mint I didn't use docker for fuxa.

Ok now I'll try to follow your suggestions

Thanks

nex2000, 2024-08-26:
I followed your advice, installed the copied driver .ini under etc

but I get an error in the connection

[image]

MatthewReed303, 2024-08-26:
@nex2000 sorry you also need to install,

`apt-get update && apt-get install -y unixodbc unixodbc-dev
`
In you terminal run the following and see if it connects,

`sudo myodbc-installer -s -a -c2 -n "test" -t "DRIVER=MySQL;SERVER=YourIP;PORT=3306;DATABASE=testDB;UID=User;PWD=MyPass"`

Then

`sudo isql test`

You can then test the connection and your SQL commands from the terminal 

If it doesn't connect you need to debug that first or it will not work in Fuxa as the node-odbc package uses unixodbc

nex2000, 2024-08-27:
Unfortunately using the ODBC driver installation script failed to install and security errors occurred.

`root@osboxes:/home/osboxes/Downloads/FUXA-master/odbc# ./install_odbc_drivers.sh
Get:1 http://repo.mysql.com/apt/debian buster InRelease [22.1 kB]
Hit:2 http://security.ubuntu.com/ubuntu jammy-security InRelease               
Hit:3 https://deb.nodesource.com/node_14.x jammy InRelease                     
Hit:4 http://archive.ubuntu.com/ubuntu jammy InRelease              
Ign:5 http://packages.linuxmint.com victoria InRelease
Hit:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease
Hit:7 http://packages.linuxmint.com victoria Release
Hit:8 http://archive.ubuntu.com/ubuntu jammy-backports InRelease
Err:1 http://repo.mysql.com/apt/debian buster InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B7B3B788A8D3785C
Reading package lists... Done
W: GPG error: http://repo.mysql.com/apt/debian buster InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY B7B3B788A8D3785C
E: The repository 'http://repo.mysql.com/apt/debian buster InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
Unsupported Debian version: 21.2
`
Regards

nex2000, 2024-08-27:
ok now I use docker 

Thanks a lot

nex2000, 2024-08-27:
Hello I tried again and now I don't get any errors from fuxa just the usual error that can be ignored.

Could you show me where to put a select to try a data extraction.

Thanks

nex2000, 2024-08-27:
Finally I did it without Docker I modified the ODBC drivers installation file.

Now I need to know how to test a select.

Thanks

[image]

MatthewReed303, 2024-08-27:
@nex2000 create a server side script in Fuxa

First you need to get ODBC connection via devices, in my example it's named postgreSQL
```
// Initialize the device, sane name as connection
let myDevice = await $getDevice('postgreSQL', true);
```
Read data from the DB
```
let result = await myDevice.pool.query(`SELECT * FROM "testTable"`);
console.log(JSON.stringify(result));
```
Use the test button in the script interface and it should return the results and print to the console:

[image]

nex2000, 2024-08-28:
I get a syntax error on the select line.

Is this a true or false error?

`let myDevice = await $getDevice('Astuto', true);
let result = await myDevice.pool.query('SELECT * FROM "pointvalues"');
console.log(JSON.stringify(result));

MatthewReed303, 2024-08-28:
Try with your DB schema 
```
let result = await myDevice.pool.query(`SELECT * FROM "your-schema"."your-table"`);
```
It will most likely be your query syntax incorrect, try with and without quotes for your table etc

nex2000, 2024-08-28:
Ok now work fine the right siyntax is 👍 

let result = await myDevice.pool.query('SELECT * FROM table_name');

Thanks a lot Mattew

nex2000, 2024-08-29:
Hello 
I created a select that extracts my value from the db and it works fine.

I create a script with interval 20 sec, and I create a numeric tag called TEST

the script has these lines I do not put the select because it is too long.
but its value is inside the RESULT variable
*******
$setTag('t_4ad60d22-17f24f23' /* FUXA Server - TEST */, ),result;
console.log(JSON.stringify(result));
********

The TEST tag does not modify the value, where am I wrong?

MatthewReed303, 2024-08-29:
@nex2000 please explain what you are trying to achieve, are your trying to store SQL results into a tag every 20 sec? Please provide the complete script as it does not make sense

nex2000, 2024-08-29:
Here you are

`let myDevice = await $getDevice('ASTUTO', true);
let result = await myDevice.pool.query('SELECT pointValues.pointValue From dataPoints, pointValues WHERE dataPoints.xid="DP_339105" and dataPoints.id=pointValues.dataPointId order BY pointValues.ts DESC LIMIT 1');
$setTag('t_4ad60d22-17f24f23' /* FUXA Server - TEST */,value );
console.log(JSON.stringify(result));`

The select work fine but my tag not show value

MatthewReed303, 2024-08-29:
That's because the result will not be returning a number most likely. What is the contents of result in console.log?

nex2000, 2024-08-29:
>[{"pointValue":504.9876594691438}]
>"Script OK: script_1"

MatthewReed303, 2024-08-29:
Note your result value is not a number and will not work, you need to extract the number from the result or change the tag to a string type and should display the result

nex2000, 2024-08-29:
Ok can you tell me what language is so convert the string in numbers.

WuBingo12, 2024-08-29:
> @nex2000 create a server side script in Fuxa
>
> First you need to get ODBC connection via devices, in my example it's named postgreSQL
>
> ```
> // Initialize the device, sane name as connection
> let [...]

Can you publish the complete content of your picture? This can be used as a complete ODBC display example. Thank you very much.

MatthewReed303, 2024-08-29:
@WuBingo12 please see the new Wiki Page for full example [ODBC Wiki](https://github.com/frangoteam/FUXA/wiki/HowTo-ODBC)

closed 2024-10-27 by unocelli