frangoteam/FUXA #1207: What's the usage of newly added system function?

IonuDragomir1, 2024-05-28:
Hi,

What's the usage of the following system functions?
1. $invokeObject('Object name', 'function name', [Parameter])
2. $runServerScript('Server script name', [Parameter])
3. $getTagDaqSettings(TagID)
Here I have tried:
console.log("getTagDaqSettings: " + $getTagDaqSettings('t_fbb6a06f-366a43de' /* modbus - t1 */));
What I get is: 
Console
>getTagDaqSettings: [object Object]
>"Script OK: script_1"
4. $setTagDaqSettings(TagID, DaqSettings)

unocelli, 2024-06-02:
Hi, documentation is currently missing.

- $invokeObject is Client side only and you can call a GUI Controls function, to do that you have to look in code witch function are available
- $runServerScript you can call from Client a script that run on server
- $getTagDaqSettings get the Tag configurations like DAQ interval...used to modify DAQ configuration in runtime
- $setTagDaqSettings set the Tag configurations like DAQ interval...used to modify DAQ configuration in runtime

IonuDragomir1, 2024-06-06:
This is what I get for $getTagDaqSettings:

[image]

Is it OK? t_9475b5e1-5bab4b04 is a tag which is archived at each 60 seconds.

unocelli, 2024-06-07:
Hi,  the console on the script don't work as expected, to see all the attributes of the object you have to convert it into a string `console.log(JSON.stringify(yourObject));`

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli

jyb1975, 2024-10-06:
Hi,

$runServerScript is client API, which is not working in script editor if you click button 'Run TEST'. I took couple of days to trace source and noticed script editor always post the script code to server even though script mode is CLIENT. Hence the exception of '$runServerScript is not defined...' will be thrown in msm-scripts.js. However, runTestScript() does not catch the exception like runScript(). The script editor can't get any exception message.

BTW, $runServerScript is a amazing API and working fine in UI event handler. I'm looking forward more APIs.