frangoteam/FUXA #1213: [FEATURE] New Ethernet/IP client - testers wanted

greg9504, 2024-06-01:
@AlexCastro-Prog
@cjmccoy2
@ZhuNathan

**Describe the feature**
I've integrated the ST-ethernet-ip node library into FUXA.  This library supports Ethernet/IP IO implicit messaging, generic explicit messages and proprietary Rockwell symbolic tags (tags by name).   See here for a description at a high level: https://library.automationdirect.com/ethernetip-implicit-vs-explicit-messaging/ 

Why?
The current Ethernet/IP plug-in is not really Ethernet/IP, it is the PCCC protocol (AB) which uses Ethernet/IP as the transport layer.  There are so many more devices which use generic Ethernet/IP.  This plugin allows access to them, as well as having support for Rockwell proprietary symbolic tags.

**Additional context**
How to use it:
Add a new device, and select GenericEthernetIP as the type:
[image]

Then click on the browse/search button.  This will search the local network for Ethernet/IP clients.  

[image]

[image]

This waits for 6 seconds for responses, any  found devices will be in the dropdown list.  In this case two were found, a Banner Engineering safety controller and a Rockwell Micro850 PLC (simulator)
[image]

Select the device you want to connect to.
[image]

Consult the device manuals.  If the device supports implicit messaging (sometimes referred to as IO Messages), you configure those here.  Later you'll configure tags that reference these messages.

If the device supports IO messages, click the plus button to add a message.  A device can support multiple messages.  So you can define more than one.  When you click the add button (or edit one you've previously added), the following dialog will appear:
[image]

All this information will be found in the device manuals.  It is important that all sizes be entered in BYTES!  Note most devices will use 2 byte words, so if the size is given in words, multiply by 2.  

IO messages are sent over UDP at a predefined rate.  That is the Request Packet Interval (RPI).  This is independent of the Polling rate defined in the device setup tab.  It is the rate specific for this IO message.  Most devices will support RPI less than 100ms, however remember that FUXA is a client/server application.  Setting this very low will only most likely not have the affect you were hoping for.  

After entering the IO modules, enable the device in the device setup panel and then close the setup panel.  If you have configured IO messages, the pluin will start to exchange those messages with the device.  Otherwise the plugin will poll the device for it's ID information.  

If you are using Wireshark, use this filter:
ip.addr==192.168.1.160&&(tcp.port==44818||udp.port==2222)

Replace the ip address with the ip address of your device.  This filter will show all Ethernet/IP TCP (port 44818) and IO messages (udp port 2222).

If running debug in Visual Studio Code, the debug console for the server will look like:

```
2024-06-01T22:44:13.740Z [INF] 	'FUXA' start
2024-06-01T22:44:13.746Z [INF] 	'XS26CP' created
2024-06-01T22:44:13.750Z [INF] 	'XS26CP' start
2024-06-01T22:44:13.752Z [DBG]  	'XS26CP' try to connect 192.168.1.160:44818
2024-06-01T22:44:13.752Z [DBG]  	XS26CP disconnectAll!!!!!!
2024-06-01T22:44:13.761Z [INF] 	'FUXA' restored 0/0 values
2024-06-01T22:44:13.764Z [INF] 	'XS26CP' restored 0/0 values
2024-06-01T22:44:13.800Z [DBG]  	XS26CP ethernet/ip explicit/symbolic conn is now connected!!!!
2024-06-01T22:44:13.801Z [DBG]  	XS26CP Make IO Connections
2024-06-01T22:44:13.802Z [DBG]  	XS26CP globalIOScanner undefined, creating new and binding to UDP socket
Bind successful
server listening 0.0.0.0:2222
2024-06-01T22:44:13.805Z [DBG]  	XS26CP globalIOScanner bind successful
2024-06-01T22:44:13.847Z [DBG]  	XS26CP in waitingForIOConnections, numberOfIOScannersWaitingToConnect: 1, connectionAttempts: 20
2024-06-01T22:44:14.361Z [DBG]  	XS26CP in waitingForIOConnections, numberOfIOScannersWaitingToConnect: 1, connectionAttempts: 19
2024-06-01T22:44:14.864Z [DBG]  	XS26CP in waitingForIOConnections, numberOfIOScannersWaitingToConnect: 1, connectionAttempts: 18
2024-06-01T22:44:15.369Z [DBG]  	XS26CP in waitingForIOConnections, numberOfIOScannersWaitingToConnect: 1, connectionAttempts: 17
2024-06-01T22:44:15.877Z [DBG]  	XS26CP in waitingForIOConnections, numberOfIOScannersWaitingToConnect: 0, connectionAttempts: 16
2024-06-01T22:44:15.878Z [DBG]  	XS26CP io connections finished connecting
2024-06-01T22:44:15.878Z [DBG]  	'XS26CP' connected!
```

Setting up tags
There are 3 types of tags:
IO implicit tags - These point to data within IO messages
Explicit message tags - These use Ethernet/IP explicit messages to exchange data with the device
Rockwell Symbolic tags - Browse a Rockwell PLCs or type in the tag name to exchange data with a Rockwell PLC.

Add a new tag the usual way.  The dialog that comes up will look like:
[image]

In the case above Symbolic tags are not supported, you'll see a message to indicate this.   If we try a device that does support symbolic tags, the device must be "enabled" in the device setup tab.  If it is not, you will see this message telling you to enable the device:
[image]

Here is an example of a device that supports symbolic tags and is enabled:
[image]

The list of tags is a tree view.  There will always be a "Global Tags" view, this represents the global variables tags on the Rockwell PLC.  Additional program tags will be shown, a branch for each program.  

NOTE: For Micro800 devices, only Global variable tags are supported (this is a device limitation, for both real and simulator).  As I've only an Micro800 simulator to test with, I have not tested program tags. Please report back..   In some cases not all global tags will be returned by the device (this may or may not be a limitation of the simulator).  You can always manually enter the tag name in the Address field.  

When you select a tag from the list, the Address and Type field are filed in.  If you do not see the tag in the list, enter the name manually in the Address field and set the Type.

This is what the tags look like in the Rockwell Connected Components Workbench:

[image]

Here are some examples of Symbolic tags created from the above PLC:
(digital output 2)
[image]

(global variable named "Greg" of type string)
[image]

(global variable named "GregArray", index 3 array indexes are 1 based!, array of unsiged bytes)
[image]

In CCW this looks like:
[image]

continued in reply

greg9504, 2024-06-02:
And here are the values:
[image]

How to define tag that references an IO message
Select Tag Type as Assembly I/O
[image]

When you do, there will be a drop down list of IO messages/tables to select from.  Select one that you configured on the device setup panel previously.

You then need to define if this is a single bit or a 16bit word value that you want to read from the IO message and the offset of that bit/word.  The offsets are 0 based.  A bit value will have both a byte offset and a bit offset into the byte (0..7).

You then specify if tag will be used to read or write the value.

Explicit Messages
You can send Ethernet Explicit messages  - message defined by Class/Instance/Attribute using getAttributeSingle/setAttributeSingle Ethernet/IP messages - to get and set values on the device.
[image]

Refer to your device manuals for explicit message Class/Instance/Attribute values.  Enter the values, then chose whether this tag will be used for getting (getAttributeSingle) and setting the value or setting only (setAttributeSingle).  There is a text field where you can enter HEX formatted (2 character, no 0x, spaces OK) string of bytes that should be appended to the message.  This is useful if your device requires additional information beyond the Class/Instance/Attribute, otherwise leave this blank. 

That covers using the plugin.

Notes:
I have tested with:
Banner Engineering XS26 Safety Controller (IO and Explicit get messages)
SEW MDX61B VFD (IO and Explicit get/set messages)
Rockwell Micro850 Simulator using Connected Components Workbench version 22.00 (Symbolic tags)

Looking for people to test with other devices.  Hopefully someone with a Rockwell PLC.  

If you know how to build from source you can find it here:
https://github.com/greg9504/FUXA/tree/ethernetipfull

robbudge, 2024-08-26:
I have a Linux Sandbox, a CompactLogix plus other EthernetIP Devices (Symbolic Tags) all on a closed VPN Network if need any devices to play with.
I am slammed with my day job at the moment , but can make these devices and network available if needed

AlexCastro-Prog, 2024-08-26:
Hello there,

Could you explain what's the issue?

Thanks,
Alex Castro

robbudge, 2024-08-26:
No issue, but if you’re looking for real hardware especially Rockwell hardware for testing I can make some available.
I have a number of isolated lab rigs for testing.

With some luck, I might be able to run some load testing in a month or so.

Sent from IPhone

greg9504, 2024-08-26:
@AlexCastro-Prog 
I think I explained the issue as well as I can in the original post above.  
@robbudge 
Thanks for the offer of the test clients.  I need to update the branch with the latest from main.  I'll be in touch to see if we can work out something.
Greg.

jpomo, 2025-02-26:
@greg9504 I can test with a CompactLogix controller. Is your fork still active?

unocelli, 2025-02-27:
Hi, sorry but frangoteam don't have CompactLogix controller, so we can't test it

robbudge, 2025-02-27:
I’m slammed so don’t have time to play but I have a compact logix that I can expose via a port forward if you have the software and time

Best regards,
Robert Budgen
System Engineer Technical Solutions Expert

Bürkert Fluid Control Systems
Burkert Contromatic Inc.
5002 South Service Road
Burlington ON  L7L 5Y7
Cell:  (905) 630-0694
Phone: (905) 632-3033
Fax:     (905) 632-3833
***@***.******@***.***>
www.burkert.ca<http://www.burkert.ca/>

Sent from IPhone

unocelli, 2025-05-15:
@greg9504 Sorry for the delay — I’m really impressed with your work. Do you think this could be integrated into the official version?

jpomo, 2025-05-15:
> [@jpomo](https://github.com/jpomo) If by active, do you mean I've kept up to date with FUXA main, no I haven't. So if there are other features in FUXA you are looking for they may not be in the fork. I was hoping this would have received some interest from the FUXA maintainers ([@unocelli](https://github.com/unocelli) ) but there didn't appear to be any. It was a lot of work getting it integrated. If you have the skills to build the fork, please give it a try with your CompactLogix controller. I will then look at bring the fork up to date with FUXA main. Greg.

I appreciate your reply - Since then I have tried (without success) to build the fork but this is because of my lack of experience, not the fork itself. If you do choose to bring it up-to-date, I will make time to test it as I have Micrologix and CompactLogix controllers available.

jpomo, 2025-05-15:
@greg9504

    [@jpomo](https://github.com/jpomo) If by active, do you mean I've kept up to date with FUXA main, no I haven't. So if there are other features in FUXA you are looking for they may not be in the fork. I was hoping this would have received some interest from the FUXA maintainers ([@unocelli](https://github.com/unocelli) ) but there didn't appear to be any. It was a lot of work getting it integrated. If you have the skills to build the fork, please give it a try with your CompactLogix controller. I will then look at bring the fork up to date with FUXA main. Greg.

I appreciate your reply - Since then I have tried (without success) to build the fork but this is because of my lack of experience, not the fork itself. If you do choose to bring it up-to-date, like @unocelli mentioned just now, I will make time to test it as I have CompactLogix controllers and other EtherNet/IP devices available.

greg9504, 2025-05-15:
> Do you think this could be integrated into the official version?

Yes, I can look at bringing it up to date.  I know when I originally wrote it, the plugin interface had changed, and this was written to the new interface.  Hopefully it hasn't changed much since :)

> will make time to test it as I have CompactLogix controllers and other EtherNet/IP devices available.

OK I will work on it in the next few days.  

I had originally wrote the plugin to work with a personal project - ABB industrial robot fitted with a milling spindle to work stone.  The spindle uses a SEW VFD, and I have a safety PLC, both of these devices talk Ethernet/IP.  I'm back working on the project, so the timing is good.

unocelli, 2025-05-18:
@greg9504 Great! let me know if I can help you with the frontend, maybe you can do the PR and I can look at where the conflicts are.

greg9504, 2025-05-19:
@unocelli OK I have done a rebase to master on my branch so it is up to date with master.  I've done some minimal testing and am able to communicate via Ethernet/IP.  So it looks like the rebase has worked.  I will do some more testing then do a PR.  For now, code is here:
https://github.com/greg9504/FUXA/tree/ethernetipfull

jpomo, 2025-05-22:
> [@unocelli](https://github.com/unocelli) OK I have done a rebase to master on my branch so it is up to date with master. I've done some minimal testing and am able to communicate via Ethernet/IP. So [...]

I got this built and running. Here is some initial feedback:

1. Device Searched worked well and symbolic browsing on the CompactLogix PLC works: 

[image]

[image]

2. I found that 'nested' tags are not browsable. For instance, for local inputs, the highest level of the tag is visible in the browser, but no sub-tags (this may be expected behavior?). Specifying the exact tag address works just fine.

  
[image]

3. At least in my build, there's an issue displaying tag values. However they display correctly in the console:

[image]

[image]

4. Program-scoped tags appear to not add correctly. The Tag browser can see the Program list in the PLC however the values are null:

[image]

[image]

greg9504, 2025-05-22:
Hi @jpomo, thanks for taking a look.  
For 3) I am seeing the same thing using the Micro800 simulator.  Ironically I think it's because I haven't updated the Ethernet/IP code to cope with another feature I worked on (Tag function scaling).  I should be able to fix this up soon.
For 2) Is it possible to show me a screen shot of how the variable Local:1:I: and Local:1:I.Data.0 is setup in the PLC software?
For 4) I'll have to get back to you.

greg9504, 2025-05-23:
@jpomo , OK I've fixed 3).  You'll need to update and and stop/start the server.  Please give it a try.

greg9504, 2025-05-23:
@jpomo  OK I think I may have fixed 4).  I wasn't passing the configured program to the method that creates the tag to send when reading values.  The software simulator for Micro800 doesn't return program variables, so I can't test it.  If you can give it another try that would be great.

jpomo, 2025-05-23:
> [@jpomo](https://github.com/jpomo) OK I think I may have fixed 4). I wasn't passing the configured program to the method that creates the tag to send when reading values. The software simulator for [...]

Thanks for the quick reply! I see that 3 and 4 (programmed-scoped) are resolved on my end, now. Regarding 2). Local:1:I is referencing discrete IO on the PLC I am using. Local:1:I.Data.[x] has  16 bits. 

[image]

However, the structure is maybe more like a User Defined Tag (UDT). There is Local1:I.Data (16bits, INT) as well as .Fault (32 bits, DINT) and .Readback (16 bits, INT). 

Those local inputs are not the only tags with nested data types. For instance, a "TIMER" is a collection of DINTs and BOOLs

[image]

User defined tags may be a collection (an array) of (I think) any of these types, e.g. timers, bools, DINTs, INTs, Strings, etc.

[image]

At the moment, all of these are accessible if you know the exact address for the tag, but are not populated in the browser.

Thanks again for your work on this. I will carve out some time next week to test with other Ethernet/IP devices as well.

jpomo, 2025-05-23:
> [@jpomo](https://github.com/jpomo) OK I think I may have fixed 4). I wasn't passing the configured program to the method that creates the tag to send when reading values. The software simulator for [...]

> [@jpomo](https://github.com/jpomo) OK I think I may have fixed 4). I wasn't passing the configured program to the method that creates the tag to send when reading values. The software simulator for [...]

Thanks for the quick reply! I see that 3 and 4 (programmed-scoped) are resolved on my end, now. Regarding 2). Local:1:I is referencing discrete IO on the PLC I am using. Local:1:I.Data.[x] has  16 bits. 

[image]

However, the structure is maybe more like a User Defined Tag (UDT). There is Local1:I.Data (16bits, INT) as well as .Fault (32 bits, DINT) and .Readback (16 bits, INT). 

Those local inputs are not the only tags with nested data types. For instance, a "TIMER" is a collection of DINTs and BOOLs

[image]

User defined tags may be a collection (an array) of (I think) any of these types, e.g. timers, bools, DINTs, INTs, Strings, etc.

[image]

At the moment, all of these are accessible if you know the exact address for the tag, but are not populated in the browser.

Thanks again for your work on this. I will carve out some time next week to test with other Ethernet/IP devices as well.

greg9504, 2025-05-23:
@jpomo I use another toolkit for the lower level Ethernet/IP.  It returns the list of symbolic tags.  I could be that the api call to get the list has that extra information and I'm ignoring it, or the device is only giving back the top level tag info.  If I gave you a node project that dumps the returned tag info, can you run it?  You would open it in Visual Studio Code, run two commands (npm install, npm start).  I think you are doing something similar now for Fuxa, as you can see the debug output.
Thanks!

jpomo, 2025-05-23:
> [@jpomo](https://github.com/jpomo) I use another toolkit for the lower level Ethernet/IP. It returns the list of symbolic tags. It could be that the api call to get the list has that extra information and I'm ignoring it, or the device is only giving back the top level tag info. If I gave you a node project that dumps the returned tag info, can you run it? You would open it in Visual Studio Code, run two commands (npm install, npm start). I think you are doing something similar now for Fuxa, as you can see the debug output. Thanks!

Sure, I can give it a try.

carlinhost, 2025-06-25:
Fantastic work! I have several Micro800 PLCs running, and this could be a great alternative to relying on their limited Modbus TCP support. Looking forward to testing it!

greg9504, 2025-07-20:
@carlinhost I believe I ran into this problem as well.  I don't have access at the moment to my FUXA build... what commands did you use to build it?  Try the debug instructions.  I'll take a look tonight.

carlinhost, 2025-07-20:
@greg9504 
1) Git clone your ethernetip branch
2) Changed /package.json and /server/package.json from "github:greg9504/ST-node-ethernet-ip#Fuxa" to "st-ethernet-ip": "2.7.2"
3) Build this docker file attached

[Dockerfile.txt](https://github.com/user-attachments/files/21337518/Dockerfile.txt)

greg9504, 2025-07-28:
@carlinhost sorry, was busy this week, should be able to look at it this week.  One thing, is you need to build with the 
"st-ethernet-ip": "github:greg9504/ST-node-ethernet-ip#Fuxa"
or you will run into problems with connections.  I made some changes to the original st-ethernet-ip, and not all have been merged back in.

However that is not what is stopping you from seeing the plugin.  

I know I ran into the same problem at one point.  If you can, try starting the VS Code debug processes.

HebigT, 2025-07-30:
> [@greg9504](https://github.com/greg9504) couldn't find the reason GenericEthernetIP is not on dropdown list of new Devices

@carlinhost I had a similar issue where it would build successfully but the plugin never showed up. I followed these exact steps for building and it worked for me. I'm new to this so I can't explain why this works at the moment, but I think this properly installs all modules as without these steps I also had errors related to missing modules when trying to start FUXA:

I have this version of Node installed:
v18.20.8

Steps:
1. git clone --branch ethernetipfull https://github.com/greg9504/FUXA.git FUXA-greg9504
2. cd FUXA-greg9504
3. npm install
4. cd client
5. npm install
6. npm run build
7. cd ..
8. cd server
9.  npm install
10. npm start (must be in server folder to start, this differs from the FUXA master, and I'm not sure why)

This is what worked for me.

carlinhost, 2025-07-31:
@HebigT It worked! Maybe I was missing "npm run build"
Thanks!

greg9504, 2025-07-31:
@HebigT @carlinhost great that you got it sorted.  As jpomo pointed out structures are currently not supported.  I have to take another look, but off the top of my head I do not believe FUXA has a way to deal with structured tags, even if they were reported by the plugin.  But I will look further into it.

jpomo, 2025-07-31:
@greg9504 Just a quick update. I've been using this branch successfully to communicate with a CompactLogix controller. I attempted with a Banner safety controller but it failed to connect. I will spend some more time on this. Connection to the AB PLC has been reliable.

carlinhost, 2025-08-01:
@greg9504 it seems that st-ethernet-ip does not support all tag.type values. I'm unable to write to a WORD variable on a Micro850 controller.
Here's the error message I get:
2025-08-01T02:15:40.767Z [ERR] micro850 'SP_HZ_M7' setValue error! Error: Unrecognized Type to Write to Controller: 209

read is working fine!

greg9504, 2025-08-03:
@carlinhost How have you defined the tag in FUXA?  Can you show me a screen shot, and how it is defined on the Micro850.  It seems that it must be sent to FUXA as something other than WORD, because the read does not specifically handle WORD either.  WORD is defined as a bit string of 16 bits.

RobertMiller-KR2, 2025-12-26:
Hey @unocelli  will this be merged into the Pro Version?

greg9504, 2025-12-30:
@kasper219 It does look like the typescript compile of st-ethernet-ip (the underlying lib used by the plugin) is failing.  But nothing jumps out to me why.  Can run again and attach the referenced log files?

greg9504, 2025-12-31:
```
79 http fetch GET 200 https://codeload.github.com/greg9504/ST-node-ethernet-ip/tar.gz/14d0afc6aeea9caa96b9097b25b7cd56b67f6495 606ms (cache revalidated)
80 timing reify:rollback:createSparse Completed in 42ms
81 timing reify:rollback:retireShallow Completed in 0ms
82 timing command:install Completed in 25188ms
83 verbose stack Error: git dep preparation failed
83 verbose stack     at ChildProcess.<anonymous> (/usr/share/nodejs/@npmcli/promise-spawn/lib/index.js:53:27)
83 verbose stack     at ChildProcess.emit (node:events:524:28)
83 verbose stack     at maybeClose (node:internal/child_process:1104:16)
83 verbose stack     at ChildProcess._handle.onexit (node:internal/child_process:304:5)
84 verbose cwd /home/kasper/FUXA/FUXA
85 verbose Linux 6.12.47+rpt-rpi-v8
86 verbose node v20.19.2
87 verbose npm  v9.2.0
88 error code 1
89 error git dep preparation failed
90 error command /usr/bin/node /usr/share/nodejs/npm/bin/npm-cli.js install --force --cache=/home/kasper/.npm --prefer-offline=false --prefer-online=false --offline=false --no-progress --no-save --no-audit --include=dev --include=peer --include=optional --no-package-lock-only --no-dry-run
91 error > st-ethernet-ip@2.7.2 prepare
91 error > npm run build
91 error
91 error
91 error > st-ethernet-ip@2.7.2 build
91 error > tsc
```

@kasper219 I'm not sure what is going on, but my guess is that the tsc command is being run in the /home/kasper/FUXA/FUXA directory instead of /home/kasper/FUXA/FUXA/server/node_modules/st-ethernet-ip

Because the output from tsc is as if it didn't find a tsconfig.json file.  

However I just notices the file is named TSConfig.json, perhaps the case sensitivity on linux is the problem.  I'm not sure of an easy way to fix this when you build.  If after failing the  /home/kasper/FUXA/FUXA/server/node_modules/st-ethernet-ip directory exists, rename to all lower case tsconfig.json and try the install again.

greg9504, 2025-12-31:
@kasper219 Yes I can fix it. It will still work on windows.   When I said I wasn't sure of an easy way to fix it, I meant without me having to publish a build (which I've forgotten how :) ).   The st-ethernet-ip isn't my project, I forked it to fix a few things.  However I just checked and it looks like the owner has made some updates. I'll open ticket for this bug (tsconfig.json should always be lowercase) and see about merging any changes I've made that are not in the main repo, then we won't need my fork. Thanks for tracking that down.
Greg.

kasper219, 2025-12-31:
@greg9504 I went ahead and made a fork to test if it worked, and I have had success with linux and docker. I will add the docker file that I am using currently to run (I haven't tested all of the functions yet, so your mileage may vary). I will put the info down below in case someone needs to run it this way.

```
FROM node:18-bookworm

RUN apt-get update && apt-get install -y dos2unix

WORKDIR /usr/src/app
# Using the fork that fixes the issue of TSConfig.conf found in the ST-ethernet-ip library in linux environments
RUN git clone --branch ethernetipfull https://github.com/kasper219/FUXA.git FUXA-kasper219

RUN mv FUXA-kasper219 FUXA

# Install build dependencies for node-odbc
RUN apt-get update && apt-get install -y build-essential unixodbc unixodbc-dev

# Convert the script to Unix format and make it executable
RUN dos2unix FUXA/odbc/install_odbc_drivers.sh && chmod +x FUXA/odbc/install_odbc_drivers.sh

WORKDIR /usr/src/app/FUXA/odbc
RUN ./install_odbc_drivers.sh

# Change working directory
WORKDIR /usr/src/app

# Copy odbcinst.ini to /etc
RUN cp FUXA/odbc/odbcinst.ini /etc/odbcinst.ini

#install client
WORKDIR /usr/src/app/FUXA/client
RUN npm install
RUN npm run build

# Install Fuxa server
WORKDIR /usr/src/app/FUXA/server
RUN npm install

# Add project files
ADD . /usr/src/app/FUXA

# Set working directory
WORKDIR /usr/src/app/FUXA/server

# Expose port
EXPOSE 1881

# Start the server
CMD [ "npm", "start" ]

```

Save the above to FUXA_Dockerfile.txt

Run the following command on linux to generate the docker image

```
docker build --no-cache -t fuxa_gip:latest -f FUXA_dockerfile.txt .
```

It can then be started with the following or using different options
```
docker run --name fuxa-kasper -d -p 1881:1881 -v fuxa_appdata:/usr/src/app/FUXA/server/_appdata -v fuxa_db:/usr/src/app/FUXA/server/_db -v fuxa_logs:/usr/src/app/FUXA/server/_logs -v fuxa_images:/usr/src/app/FUXA/server/_images fuxa_gip:latest
```

unocelli, 2026-01-30:
@greg9504 @kasper219 Sorry, I may have missed this. What is the current status of this feature?

kasper219, 2026-02-03:
@unocelli @greg9504 From my experience the only bug/issue that I have found was the tsconfig.json file having case sensitivity issues in linux from the origional repo for st-ethernet-ip. Greg9504 opened a ticket with that repo's owner.

kasper219, 2026-04-03:
@greg9504 

Not sure if this will help or not, but I did finally get it to force an error on the web console with the dialog when selecting genericEthernetIP in the device connection property screen. So I am assuming its just a mis-match for moving over from legacy-dialog to dialog since it was depreciated out of the library and wouldn't compile.

` Example
import { MAT_LEGACY_DIALOG_DATA as MAT_DIALOG_DATA, MatLegacyDialogRef as MatDialogRef } from '@angular/material/legacy-dialog';

switched to

import { MAT_DIALOG_DATA, MatDialogRef } from '@angular/material/dialog';

`

[image]

`app.component.ts:53 FUXA v1.3.0-2778
core.mjs:31084 Angular is running in development mode.
fuxa-editor.min.js:21 editor v2.9.10-167
core.mjs:7404 ERROR RuntimeError: NG0300: Multiple components match node with tagname button: MatIconButton and MatMiniFabButton. Find more at https://angular.dev/errors/NG0300
    at throwMultipleComponentError (core.mjs:10794:11)
    at findDirectiveDefMatches (core.mjs:12866:29)
    at resolveDirectives (core.mjs:12666:29)
    at elementStartFirstCreatePass (core.mjs:24301:5)
    at ɵɵelementStart (core.mjs:24337:11)
    at DevicePropertyComponent_div_3_div_11_div_14_Template (template.html:600:21)
    at executeTemplate (core.mjs:12159:9)
    at renderView (core.mjs:13366:13)
    at createAndRenderEmbeddedLView (core.mjs:13436:9)
    at TemplateRef.createEmbeddedViewImpl (core.mjs:14389:31)
handleError @ core.mjs:7404
core.mjs:7404 ERROR TypeError: Cannot read properties of null (reading 'flags')
    at isDirectiveHost (core.mjs:3877:19)
    at ɵɵelementStart (core.mjs:24341:27)
    at DevicePropertyComponent_div_3_div_11_div_14_Template (template.html:601:25)
    at executeTemplate (core.mjs:12159:9)
    at renderView (core.mjs:13366:13)
    at createAndRenderEmbeddedLView (core.mjs:13436:9)
    at TemplateRef.createEmbeddedViewImpl (core.mjs:14389:31)
    at ViewContainerRef.createEmbeddedView (core.mjs:17109:37)
    at SwitchView.create (common.mjs:3569:32)
    at SwitchView.enforceState (common.mjs:3577:18)
handleError @ core.mjs:7404
(anonymous) @ core.mjs:7429
invoke @ zone.js:369
run @ zone.js:111
runOutsideAngular @ core.mjs:7142
(anonymous) @ core.mjs:7429
_tick @ core.mjs:32789
tick @ core.mjs:32769
(anonymous) @ core.mjs:33195
invoke @ zone.js:369
onInvoke @ core.mjs:7251
invoke @ zone.js:368
run @ zone.js:111
run @ core.mjs:7097
next @ core.mjs:33194
next @ Subscriber.js:91
_next @ Subscriber.js:60
next @ Subscriber.js:31
(anonymous) @ Subject.js:34
errorContext @ errorContext.js:19
next @ Subject.js:27
emit @ core.mjs:6789
checkStable @ core.mjs:7165
onLeave @ core.mjs:7312
onInvokeTask @ core.mjs:7245
invokeTask @ zone.js:401
runTask @ zone.js:159
invokeTask @ zone.js:483
invokeTask @ zone.js:1138
globalCallback @ zone.js:1169
globalZoneAwareCallback @ zone.js:1202
`

HebigT, 2026-04-21:
I just tested the latest branch with a CompactLogix controller with success! (commit id f24cfbe)

HebigT, 2026-04-24:
@greg9504 

OK I did a comparison - modifying the index.js file allowed both of my ethernet IP devices to reconnect, whereas they remained disconnected before and needed to be reenabled in FUXA.

kasper219, 2026-08-18:
@unocelli @greg9504 Is there anything I can or need to do to help this along in getting in the main branch?

montillaex, 2026-08-31:
> [@unocelli](https://github.com/unocelli) [@greg9504](https://github.com/greg9504) Is there anything I can or need to do to help this along in getting in the main branch?

Hi everyone, this will be a great feature, because Ethernet-IP is the one most protocol used in the industrial automation around the world. Waiting for to Feature status.

HebigT, 2026-08-31:
@unocelli I too am interested in having this feature. I work almost exclusively with Allen Bradley controllers.

montillaex, 2026-09-28:
Maybe the payment version can included this communication driver? I'll pay for it..

greg9504, 2026-09-29:
@unocelli Are you interested in this?  I think the testing done is enough to proceed with integrating into the main build.  I haven't made documentation yet,but the first post in this thread contains the information for the documentation.  What do you need me to do?
Thanks
Greg.

unocelli, 2026-09-29:
@greg9504 Yes, definitely.
I don’t remember if we already discussed how to proceed with the integration. Did you have a particular approach in mind, for example creating a PR?
Thanks Greg!

kasper219, 2026-10-06:
@greg9504 I can also confirm that I can read Global tags on a Micro850 with this as well. I have also been testing with multiple CompactLogix and have not had any issues as of yet with this library.

@greg9504 @unocelli  Let me know if there is anything I can help with in getting this into the Main branch of FUXA as it is a very nice feature to have.

Thank you @greg9504 for all the work thus far!

unocelli, 2026-10-08:
Hi @kasper219 I've created a PR for this issue. It would be great if you could give it a try and let me know if it works. #2586

kasper219, 2026-10-09:
@unocelli I should be able to test tonight and this weekend. I have a Micro850 and a CompactLogix I can test with.

robbudge, 2026-10-09:
I’ll try and set something up this weekend I have a 5590 control logix on our current project with around 70k Tags

Best regards,
Robert Budgen
System Engineer Technical Solutions Expert

Bürkert Fluid Control Systems
Burkert Contromatic Inc.
5002 South Service Road<x-apple-data-detectors://4/1>
Burlington ON  L7L 5Y7<x-apple-data-detectors://4/1>
Cell:  (905) 630-0694<tel:(905)%20630-0694>
Phone: (905) 632-3033<tel:(905)%20632-3033>
Fax:     (905) 632-3833<tel:(905)%20632-3833>
***@***.******@***.***>
www.burkert.ca<http://www.burkert.ca/>

Sent from IPhone

closed 2026-10-19 by unocelli