frangoteam/FUXA #1217: Easier ODBC installation [FEATURE]

Kapket, 2024-06-04:
Is it possible to add an ODBC installation like EthernetIP? I see that the plugin is already in the appropriate tab, but the option to add is missing.

Currently adding ODBC is quite troublesome. When I managed to put up a server using Dockerfile, I had no option to change the database connection port, but when I try to put up a server by 'npm start' and install the plugin manually, the system does not detect the installation.

MatthewReed303, 2024-06-04:
Hi @Kapket please see this issue where I describe how to use it, please note it's not in the current release and you have to use the odbc branch https://github.com/frangoteam/FUXA/issues/1204

unocelli, 2024-06-07:
Hi, thanks to @MatthewReed303 we will be able to merge on the master branch soon

closed 2024-10-27 by unocelli