frangoteam/FUXA #1219: Scripting incompatibility V1.1.18-1411  --> V1.1.19-1619

lewsut, 2024-06-06:
Hello,

I have been out of the game for a while and updated the other day, however non of my button handling (and more) scripting seems to work correctly.

I have looked through the FUXA change logs (can't see anything of interest)  and before I start digging I though I would ask if there was something obvious to you that has changed?  Example code below, Button_Flag  gets stuck as true, this always works as expected on V1.1.18

```
if ( $getTag('t_04be4540-6b5d44b2' /* FUXA Server - Button_Flag */) ==0 ){

$setTag('t_04be4540-6b5d44b2' /* FUXA Server - Button_Flag */, 1);

if ( $getTag('t_87513fab-a92b4cef' /* MQTT_Send - Pump_Set_RX */) == 1){
$setTag('t_87513fab-a92b4cef' /* MQTT_Send - Pump_Set_RX */, 0 );  
}

else if ($getTag('t_87513fab-a92b4cef' /* MQTT_Send - Pump_Set_RX */) == 0){
$setTag('t_87513fab-a92b4cef' /* MQTT_Send - Pump_Set_RX */, 1 );  
}
  setTimeout(() => { $setTag('t_04be4540-6b5d44b2' /* FUXA Server - Button_Flag */, 0); }, 500);
}
```

unocelli, 2024-06-07:
Hi, $getTag and $setTag have not been modified

lewsut, 2024-06-11:
Interesting, I'll have a dig around then, it's very odd, my scripts are broken in the new version, work as expected right away when I revert.

unocelli, 2024-06-19:
Can you share the error that you receive?

lewsut, 2024-07-10:
sorry @unocelli, I have had little time to test. It seems to be a difference in initialisation values (via scripts) of a few server tags and I can't see why yet. I'll keep digging.

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli