frangoteam/FUXA #1234: [BUG] Install freezes on Raspberry Pi Zero 2 W

abishur, 2024-06-17:
Installed Raspberry Pi OS Lite (64-Bit) on Raspberry Pi Zero 2 W.
Added the Node 14.x branch using: curl -sL https://deb.nodesource.com/setup_14.x | sudo bash -
Installed Node using sudo apt install nodejs=14.21.3-deb-1nodesource1
(I had to do the =14.21.3 to force it to install v14.  Without it even after using the script downloaded by curl it still installed v18.x)
Downloaded the latest source code zip file
Extracted zip file, entered server folder and ran npm install

It goes well until it starts doing node-gyp rebuild then I get the following error:

`Traceback (most recent call last):
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 50, in <module>
    sys.exit(gyp.script_main())
             ^^^^^^^^^^^^^^^^^
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 554, in script_main
    return main(sys.argv[1:])
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 547, in main
    return gyp_main(args)
           ^^^^^^^^^^^^^^
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 520, in gyp_main
    [generator, flat_list, targets, data] = Load(
                                            ^^^^^
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 136, in Load
    result = gyp.input.Load(build_files, default_variables, includes[:],
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 2782, in Load
    LoadTargetBuildFile(build_file, data, aux_data,
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 391, in LoadTargetBuildFile
    build_file_data = LoadOneBuildFile(build_file_path, data, aux_data,
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/input.py", line 234, in LoadOneBuildFile
    build_file_contents = open(build_file_path, 'rU').read()
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: invalid mode: 'rU' while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:400:28)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:285:12)
gyp ERR! System Linux 6.6.31+rpt-rpi-v8
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/nbc/FUXA-v.1.1.18/server/node_modules/epoll
gyp ERR! node -v v14.21.3
gyp ERR! node-gyp -v v5.1.1
gyp ERR! not ok
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.3 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm64"})

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! epoll@4.0.2 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@4.0.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/nbc/.npm/_logs/2024-06-17T16_25_50_648Z-debug.log`

I've tried running it as sudo, I've tried a previous release, not sure what I'm doing wrong to have it failing like this 🤔

abishur, 2024-06-17:
I'll give that a go.  With the docker container is it possible to add in node modules?  I was going to try and use the onoff module for use with the raspberry pi's gpio

MatthewReed303, 2024-06-17:
You can access the container directly and manually add them as a test, but if you restart docker they will be gone. You can clone the master branch and add the modules you need to the packages and then do a docker build with the added packages. 

Another way would be via python and modbus to access the GPIO and modbus in Fuxa to control them?

abishur, 2024-06-18:
Using docker got it up and running.  If I want to use the onoff module, do I add that to the package.json list in the docker container and run npm install again, or just run npm install -g onoff inside the container?

unocelli, 2025-05-04:
Closing this due to inactivity. If the issue is still relevant or unresolved, feel free to reopen.

closed 2025-05-04 by unocelli