frangoteam/FUXA #1239: install failed with some deps(xgplayer) build fail

YUHAI0, 2024-06-20:
I startclient with npm install &&npm start wich node version 14.21.3 and npm version 6.14.18 in Linux, it appear with these error:

Error: node_modules/xgplayer/es/plugins/common/optionsIcon.d.ts:17:5 - error TS2416: Property 'show' in type 'OptionsIcon' is not assignable to the same property in base type 'Plugin'.
  Type '(value?: string) => void' is not assignable to type '(value?: string) => string'.
    Type 'void' is not assignable to type 'string'.

17     show(value?: string): void;
       ~~~~

Error: node_modules/xgplayer/es/plugins/poster/index.d.ts:25:5 - error TS2416: Property 'show' in type 'Poster' is not assignable to the same property in base type 'Plugin'.
  Type '(value?: string) => void' is not assignable to type '(value?: string) => string'.
    Type 'void' is not assignable to type 'string'.

25     show(value?: string): void;
       ~~~~

Error: node_modules/xgplayer/es/plugins/progress/index.d.ts:57:5 - error TS2416: Property 'show' in type 'Progress' is not assignable to the same property in base type 'Plugin'.
  Type '(value?: string) => void' is not assignable to type '(value?: string) => string'.
    Type 'void' is not assignable to type 'string'.

57     show(value?: string): void;
       ~~~~

Error: node_modules/xgplayer/es/plugins/progressPreview/index.d.ts:65:5 - error TS2416: Property 'show' in type 'ProgressPreview' is not assignable to the same property in base type 'Plugin'.
  Type '(value?: string) => void' is not assignable to type '(value?: string) => string'.
    Type 'void' is not assignable to type 'string'.

65     show(value?: string): void;
       ~~~~

Error: node_modules/xgplayer/es/plugins/replay/index.d.ts:15:5 - error TS2416: Property 'show' in type 'Replay' is not assignable to the same property in base type 'Plugin'.
  Type '(value?: string) => void' is not assignable to type '(value?: string) => string'.
    Type 'void' is not assignable to type 'string'.

15     show(value?: string): void;
       ~~~~

Error: node_modules/xgplayer/es/plugins/start/index.d.ts:34:5 - error TS2416: Property 'show' in type 'Start' is not assignable to the same property in base type 'Plugin'.
  Type '(value?: string) => void' is not assignable to type '(value?: string) => string'.
    Type 'void' is not assignable to type 'string'.

34     show(value?: string): void;
       ~~~~

Error: node_modules/xgplayer/es/plugins/time/index.d.ts:20:5 - error TS2416: Property 'show' in type 'Time' is not assignable to the same property in base type 'Plugin'.
  Type '(value?: string) => void' is not assignable to type '(value?: string) => string'.
    Type 'void' is not assignable to type 'string'.

20     show(value?: string): void;
       ~~~~

LordNetro, 2024-06-20:
Hello!

It seems like the problem is with the TypeScript, but usually these errors are caused by conflicts or older versions... Have you tried removing both `node_modules` folder and `package-lock.json` file, and doing a fresh install? It often solves all my problems with that type of errors...

I hope this helps you!

YUHAI0, 2024-06-20:
it because the xgplayer newly version has build problem, I lock to version 3.0.14 solve this problem.

YUHAI0, 2024-06-20:
> Hello!
> 
> It seems like the problem is with the TypeScript, but usually these errors are caused by conflicts or older versions... Have you tried removing both the `node_modules` folder and `package-lock.json` file, and doing a fresh install? It often solves all my problems with that type of errors...
> 
> I hope this helps you!
> 
> EDIT: And be careful with the npm audit fix... it usually messes up all your environment.

it because the xgplayer newly version has build problem, I lock to version 3.0.14 solve this problem.

unocelli, 2024-06-21:
Hi, this issue has already been fixed

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli