frangoteam/FUXA #1240: [BUG] html-chart lable will show forever since I delete this control on svg-editor

YUHAI0, 2024-06-21:
[image]
 As shown above, the lable can't disappear since I delete the html-chart control.

YUHAI0, 2024-06-24:
> Hi, describe how to reproduce it please

OK, I focus the pointer on the html-chart until it show the label, then I delete the control use the delete key, then the label can't disappear

YUHAI0, 2024-07-02:
[image]
@unocelli
Althrough this is a small bug, but I think it's importan to user experience, if you have enough time I recommend you to check it and merge the PR that fixed this issue.

unocelli, 2024-07-12:
@YUHAI0 Thanks

closed 2024-07-12 by unocelli