frangoteam/FUXA #1243: Too many Errors in the installation of FUXA using NPM command in CMD

abdulbasit0001, 2024-06-22:
[image]

#Facing these issues while the installation of FUXA. Please tell me how to resolve it? I changed the Python version but it did not work. What else can I do?

abdulbasit0001, 2024-06-22:
There is not a single error there are too many errors I do not know how to resolve it. As soon as I use this command 
**npm install -g --unsafe-perm @frangoteam/fuxa** the errors appears.

unocelli, 2024-06-23:
Hi, try the [version](https://www.npmjs.com/package/@frangoteam/fuxa-min) without node-snap7 (Siemens S7 communication) 
`npm install -g --unsafe-perm @frangoteam/fuxa-min`

mariorl, 2024-06-23:
Hi, in my case (Windows) I'm almost sure this problem (and more concrete "cant find any visual studio installation to use") can be avoided when you install node, during that (last step), you are asked about the installation of some external dependencies (from powershell). Click yes and be patient because it takes time. Please if you get fixed in this way tell us.

abdulbasit0001, 2024-06-24:
> Hi, try the [version](https://www.npmjs.com/package/@frangoteam/fuxa-min) without node-snap7 (Siemens S7 communication) `npm install -g --unsafe-perm @frangoteam/fuxa-min`

[image]

Still it is not working, I am unable to install FUXA after attempting many attempts this method still not working.

abdulbasit0001, 2024-06-24:
Still it is not working, I am unable to install FUXA after attempting many attempts this method still not working.

abdulbasit0001, 2024-06-24:
> Hi, in my case (Windows) I'm almost sure this problem (and more concrete "cant find any visual studio installation to use") can be avoided when you install node, during that (last step), you are [...]

[image]

After attempting this method I cannot solve my issue in installation.

abdulbasit0001, 2024-06-24:
After attempting this method I cannot solve my issue in installation.

mariorl, 2024-06-24:
seems you need to install node before, since npm command is not recognized

sephi251090, 2024-06-24:
Hello, you need to install Node.js with npm first. Check your windows Path after. And, it's now better to user PowerShell in windows, instead of cdm :)

abdulbasit0001, 2024-07-01:
I have resolved my issue successfully and now FUXA has insatalled without any error. There were several issues which I am mentioning here for the help of others:

1. Install the Python latest version and the correct path in Environmental Variables (C:\Program Files\Python312)
2. Install and update Visual Studio latest version and do not forget to download the feature name Desktop development with C++
3. Install specific version of Node JS and NPM i.e Node 14.21.3 NPM 6.14.18 (Do not use latest version only this)
4. Run the command in the last to install FUXA successfully without any errors. Warnings are acceptable.

closed 2024-07-01 by abdulbasit0001