frangoteam/FUXA #1245: [FEATURE]Enable Support for Address 0 in Modbus Plugin

Hiemer23, 2024-06-24:
I am facing an issue while using the Modbus Plugin. It is not possible to enable address 0; only addresses from 1 to 65536 are allowed. However, in certain situations, we have products that already have registers at address 0, which prevents proper communication with these devices.

For example, when using the Modbus-Rtu 1 Channel 12V Relay Module Switch Input/Output RS485/TTL Communication, this device uses the following command to turn on relay 1: 01 05 00 00 FF 00 8C 3A. Since the relay 1 address is at position 0, it cannot be used due to this limitation in the Modbus Plugin.

I would like to request a solution to allow enabling address 0 so that we can communicate correctly with devices that use this address.
[image]
[image]

rvbatista, 2024-06-24:
This is a notation issue, I didn't test here, but I'm pretty sure that your address 0 is the same as classic modbus 1.

Hiemer23, 2024-06-24:
I understand your point about notation. However, I tested using address 2, and it worked correctly, showing an offset of 1 for all addresses in this case. This confirms that address 0 is the same as the classic Modbus address 1.

closed 2024-06-24 by Hiemer23