frangoteam/FUXA #1247: [FEATURE] Backup for data archive

robsori, 2024-06-25:
Hi, any SCADA software has the possibility to make a backup for the database. It should be done on demand or automatically after a time configuration. I suggest the integration of such a useful option in Fuxa.

unocelli, 2024-06-30:
Hi, we can make backup in json file.
[image]

What you mean with automatically after a time configuration?

robsori, 2024-07-01:
Hi, I think I was misunderstood. I was referring to the database backup for data and alarms. In SCADA software environments, you have the possibility to make a backup of these databases on demand or periodically by scheduling the backup action at a set time interval. Many times this backup can be done on a different data carrier to ensure the preservation of the archived data.

unocelli, 2024-07-07:
Hi, I think that a scada system should not take care of making backups, particularly if it is working. There are much more suitable tools for making backups

robsori, 2024-08-06:
Hi. I will give as an example the SCADA environment from Siemens, WinCC. It allows the creation of backups with periodic alarms at a predetermined time interval. There are customers who want to keep information with alarms for years and years and want to have the possibility to analyze these backups when they want. In addition, if the Fuxa project becomes unusable due to various factors, the possibility to recover the previous databases with alarms would be a plus.

jay-jjwu, 2024-09-04:
looking forward to this feature too!!

abuijs, 2025-05-11:
Also looking forward to this feature.