frangoteam/FUXA #1251: [BUG] Proc-Eng Shapes dont apply processValues and processAction at the same time

Delorean12DMC, 2024-06-26:
**Describe the bug**
When i use a symbol out of Proc. Eng. I can't apply process Values to define two states of the shape color and at the same time add blinking to the shape in case of an fault in the pump. It works with the normal shapes like an Rectangle, but not with the Symbols from Proc. Eng. .
It kind of has a strange behaviour. The colors i define as fill and stroke color in the process Values are dominating, when i for example remove the fill color from the symbol, it can show me the blinking, but now i dont have a status indicator anymore, just an fault state.

**To Reproduce**
Steps to reproduce the behavior:
1. Create two Variables as bool that you can use for Testing, one as a state if the pump is running or not (PumpSwitch) and another for an fault state of the pump (PumpFault)
2. Create a rectangle and any symbol from Proc. Eng.
3. Define both shapes the same way:
    - Property:
        - Tag: PumpSwitch
        - Mask Value: False [0-0]    #ffffffff        #000000ff
        - Mask Value: True [1-1]     #00ff00ff    #000000ff

    - Actions
        - Tag: PumpFault
        - Mask Value: True [1-1]    Type: Blink
        - Fill #ffffffff    Stroke #000000ff    Interval (ms) 250    Fill #ff0000ff    Stroke #000000ff

4. Define a way to switch the Values PumpSwitch and PumpFault, use a Button or real Hardware, it doesnt matter

**Expected behavior**
Both values are false, when PumpSwitch is set to true, the Proc. Eng. Symbol and Rectangle will change the fill color to Green and when disabled will change the fill color back to white, thats correct.
Now we set the PumpFault to true, the Rectangle will start blinking white/red but not the Proc. Eng. shape.
When PumpSwitch gets set to true now, the only thing that changes is that the Proc. Eng. shape sets its fill color to green, the ractangle will still be blinking.
So the rectangle has the correct behavior in that case, the blinking value overrides the property value.

Side Note: When we replace the Fill values in the Proc. Eng. shape to null (not define anything) or completely remove it, the Proc. Eng. Symbol will use the Blinking attribute.

**Environment**
 - FUXA version: 1.1.19-1621
 - Node.js version: 14.21.3
 - npm version: 6.14.18
 - Platform/OS: Windows 10 IoT Enterprise LTSC 21H2
 - Browser: Google Chrome 126.0.6478.115

Pelikan-71-57-1, 2024-07-03:
I think I have the same Problem with a slightly different description but the Same Problem

This issue here also https://github.com/frangoteam/FUXA/issues/1260

FUXA : 1.1.19 1619
Browser: Firefox
OS Server:Fedora Linux

unocelli, 2024-07-12:
Thanks for the report

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli