frangoteam/FUXA #1256: [FEATURE] Placeholder in Charts

unocelli, 2024-06-30:
### Discussed in https://github.com/frangoteam/FUXA/discussions/1230

<div type='discussions-op-text'>

<sup>Originally posted by **henjoe** June 12, 2024</sup>
Will it be possible to have a placeholder in the charts? 
Instead of selecting an specific tags here:
[image]

It would be nice if we can use Placeholder and then bind that placeholder variable to a real-tags. 

Use Cases:
 - Creating an templated graphics that has a Trend graph on it. </div>

killerbeat12, 2024-07-04:
Yes, also looking for this great feature!

unocelli, 2024-07-14:
do you have any suggestions as to how this should work?

killerbeat12, 2024-07-16:
Yes, first i think is to clarify my meaning of placeholder, a lot of SCADA systems i used have the option to give a prefix or suffix. for example if i have the following tag: Device1_Tag1, the prefix in this tag = Device1, the suffix is Tag1 both could be addressed with a placeholder for example on this way:
If i have a prefix needed for a tag i will write the following tagname to my object %DeviceName%_Tag1
With an other object event i could assign the real name like %DeviceName% = "Device1"
So you could check in your code that everything between %...% will be replaced with this string

Does this make sense?

Next to the placeholder, with the chart i am not sure if you could make this feasible but maybe with scripting?