frangoteam/FUXA #1260: [BUG]The component color cannot be displayed correctly when blink stops.

wyplzx, 2024-07-02:
**Describe the bug**
There are two fields, VALUE and ERR. 
The field of VALUE  controls the display color of components (such as rectangles), and the  field of ERR  controls the blink action of the components.
When its VALUE=0, set the color of the component to green. When its VALUE=1, set the color of the component to red. 
When its ERR=1, set the component to blink, the blink color is between yellow and blue.

**To Reproduce**
Steps to reproduce the behavior:
1. set VALUE=0 and ERR=0.   the color of the component is green, and  the component does not blink.
[image]

3. set VALUE=0 and ERR=1.  the component blinks.
[image]
[image]

5. set VALUE=1 and ERR=0.  the component stops blink, but the color of the component is green, not red.
[image]

7. See error

**Expected behavior**
at step of No.3. 
when set VALUE=1 and ERR=0.  the component stops blink, the color of the component should be red.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment**
 - FUXA version: 1.1.19   (installed in docker)  (frangoteam/fuxa:latest)
 - Node.js version: 
 - npm version:
 - Platform/OS:  centos
 - Browser:  chrome

unocelli, 2024-07-07:
Hi, can you share a project example please?

Pelikan-71-57-1, 2024-07-08:
Does this help you 
(got more or less the same Problem)
if you have Questions I will try to help

you can see the Problem in the 0_0_0_Main_Screen window
the tiles Schlafzimmer and ROBO should change colore
it was working in the Previous Version
[fuxa-project(1).json](https://github.com/user-attachments/files/16134272/fuxa-project.1.json)

wyplzx, 2024-07-09:
It uses fuxa server variables as test parameters. Because fuxa server parameters cannot modify two field variables at the same time. Reproduce using the same ERR parameter in the test demo.
For component color settings:
ERR=0: Display in green.
ERR=1: Display red and provide flashing (alternating yellow/blue colors).
Reproduction process:
1. When setting ERR=0, green is displayed. (Normal)
2. When setting ERR=1, it flashes. (Normal)
3. When setting ERR=0, stop flashing. But it shows red instead of green.
[test_fuxa-project.json](https://github.com/user-attachments/files/16137130/test_fuxa-project.json)
pictures:
(1).Initial Screen
[image]
(2). set initial value: (OUT=0, ERR=0)
[image]
(3).set ERR=1:
The shape turns the color red, then blinks.
[image]
[image]
(4).reset ERR=0:
the shape does not blink, but the color is red, not green.
[image]

unocelli, 2024-07-12:
Hi, Thanks for the report. we fix it as soon as possible

wyplzx, 2024-07-15:
hi,  @unocelli  @Pelikan-71-57-1, Thank you very much.  Good job.  Looking forward to the release of the new repaired version.

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli

wyplzx, 2024-09-05:
hi,  @unocelli
    Which version branch this bug has been merged and resolved to?
    I upgrade my environment to FUXA V.1.2.0-1815 (install by docker. )
    After checking, the fault seems still exist.
    Thank you..
[image]