frangoteam/FUXA #1261: [FEATURE] Add bool value support in actions

YUHAI0, 2024-07-03:
In production environment, we have some tag value that is bool, we need to build some actions to thoese tags, for example when one tag change to true, then the control trun to clockwise.

LordNetro, 2024-07-03:
Hi YUHAI0,

I think you can achieve that by going to the control properties of the element you want to control and creating an action:
[image]

YUHAI0, 2024-07-04:
In my version of FUXA, I don't see the Mask value option in actions, what version do you use ? @LordNetro 
[image]

LordNetro, 2024-07-04:
@YUHAI0 I use 1.1.19 but if you click on `1/0` it should change to Mask Value. 
[image]
[image]

Anyways, if you put Min: 0 Max: 0 for **false** and Min: 1 Max: 1 for **true** it should work perfectly.
[image]

YUHAI0, 2024-07-04:
@LordNetro , ok I have see that. But unfortunately it still didn't solve my problem. In fact my tag value like string, it is "true" or "false", not 1 or 0.

LordNetro, 2024-07-04:
@YUHAI0 is there any particular reason of why are you using a string to control that? couldn't you use a boolean instead? (with a boolean, the 1 or 0 works like a charm).

YUHAI0, 2024-07-04:
If I use 1 or 0, I must use something like rule engine to convert my device data "true"/"false" into 1/0, If fuxa support string value compare in action, it will be easy.

LordNetro, 2024-07-04:
@YUHAI0 I see... so the only work around and easy fix for now I would say that is a little script that constantly reads that variable and parses it to an internal boolean, it almost doesn't reduce the performance.

With an script like this one, scheduled to perform every X seconds it would work quite good.
```
if ($getTag('your_device_var_id') === "false") {
  $setTag('your_internal_boolean_id', true);
} else if ($getTag('your_device_var_id') === "false") {
  $setTag('your_internal_boolean_id', false);
}
```

unocelli, 2024-07-12:
A string "true" or "false" is not a  boolean!

LordNetro, 2024-07-16:
With that code im parsing a string to a boolean, as if the string of his device variable is "true" or "false" an internal variable will be assigned to true or false respectively, i don't know if i'm explaining myself well... It's not using a string as a boolean, is reading a string and assigning a boolean according to the value... See the code.

closed 2025-10-03 by unocelli