frangoteam/FUXA #1266: [BUG] Fuxa Electron Application not working

antwal, 2024-07-07:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

Docker Compose
```
version: "3.9"
services:
  fuxa:
    image: frangoteam/fuxa:latest
    networks:
      - blackhole
    container_name: fuxa
    volumes:
      - /mnt/storage/fuxa/appdata:/usr/src/app/FUXA/server/_appdata
      - /mnt/storage/fuxa/db:/usr/src/app/FUXA/server/_db
      - /mnt/storage/fuxa/logs:/usr/src/app/FUXA/server/_logs
      - /mnt/storage/fuxa/images:/usr/src/app/FUXA/server/_images
    ports:
      - 1881:1881
    restart: unless-stopped

networks:
  blackhole:
    name: container-net
    external: true
```

## Server:
[npm install](https://pastebin.com/Bpw0rqJa)

## Client:
[npm install](https://pastebin.com/1SGbMzbQ)
[npm run build](https://pastebin.com/C6yL6f8f)

## App
[npm install](https://pastebin.com/y6KQ2tB1)
[npm run package](https://pastebin.com/cXpzsXPz)

by default the x86_64 version is created in linux via electron-forge; I also tried creating versions for other operating systems and architectures

[npm run package (others)](https://pastebin.com/CbLARXSe)

**Expected behavior**
that the electron application worked, with the default settings described, without having to load and configure everything every time. I was hoping you could make it work like the [demo](https://github.com/frangoteam/frangoteam.github.io) so it would be lighter to use on a Raspberry Pi Zero W 2

**Screenshots**
If applicable, add screenshots to help explain your problem.

## Test Default Electron App - Linux Desktop
[image]

On Linux desktop it works but it is always empty, it doesn't load the project automatically, if I import it I have to fix something, I save and when I close and reopen the application it is completely empty.

## Test Electron App - Mac OS
[image]

## Test Electron App - Windows 11
[image]

[image]

on windows it remains blocked and does not start in both cases

## Test Electron App - Linux Arm64

I tried the application on: Raspberry Pi 4 4Gb with default image desktop and Raspberry Pi Zero W 2 with desktop xfce now I get the same error and it doesn't start

[image]

**Environment**
 - FUXA version: latest
 - Node.js version: 16.20.2 (docker image version)
 - npm version: 8.19.4 (docker image version)
 - Platform/OS: Mac OS 10.15.7 
 - Browser: Chrome 125.0.6422.142 (x86_64)

Epytus, 2024-07-08:
I have the same problem too, The electron application does not work on raspberry and windows both generating errors at startup. having compiled with node js 18 as described, I expected the electron application to work 

**To Reproduce**

Run Docker Container
`sudo docker run -it -v /home/utente/fuxa:/build debian:stable-slim /bin/bash`

Build FUXA
```
  cd /build
  echo "Clean old app"
  rm -Rf ./out
  echo "Update and Install NodeJS"
  apt update
  apt install git curl -y
  curl -sL https://deb.nodesource.com/setup_18.x | bash -
  apt update
  apt install nodejs -y
  echo "Clone FUXA"
  git clone https://github.com/frangoteam/FUXA.git
  cd /build/FUXA/server
  echo "Install Server"
  npm install
  cd /build/FUXA/client
  echo "Install Client"
  npm install
  echo "Build Client"
  npm run build
  cd /build/FUXA/app
  echo "Install App"
  npm install
  echo "Package Apps"
  npm run package -- --arch=arm64 --platform linux
  npm run package -- --arch=x64 --platform win32
  echo "Copy App Archive"
  mv ./out /build/out
  cd /build
  echo "Clean FUXA"
  rm -Rf ./FUXA
  exit
```

[Build Log](https://0bin.net/paste/zh0Y0SsF#T9g67NcLXU08R2mEzvRInibpraubkg1BNW0rDCJP4rc)

**Screenshots**

Linux Desktop (working)
[image]

Windows 11
[image]

Raspberry Desktop
[image]

**Environment**
 - FUXA version: git master
 - Node.js version: 18.20.3
 - npm version: 10.7.0

unocelli, 2024-07-25:
Hi, Thanks for the reports. 
@rikugun do you have any idea what could be the issue?

rikugun, 2024-07-29:
Hi all, Thanks for the reports.
when you build with docker , some runtime(like sqlite3) for the server is not correct, try build with native operate system.

antwal, 2024-07-29:
@rikugun, now I don't remember what the problem was in the dependencies after various tests, but anyway it couldn't compile some libraries on Mac and Raspberry, Currently I have abandoned the use of fuxa since it works well only on docker but to run it you need at least an rpi 4 with 2gb of ram and the costs are very high today. If I find the test files I'll publish them to you, anyway I remember the problem was on a couple of dependencies one in particular was odbc

antwal, 2024-07-29:
@rikugun ok, I found the old screenshots, I attach the problem

[image]

I also tried with: 
```
sudo apt-get install libsqlite3-dev
npm install sqlite3 --build-from-source --sqlite=/usr
```
but it still doesn't work

rikugun, 2024-08-01:
@antwal , I try on my nanopi. the odbc lib is not support arm64, so you can comment the deps with server package.json.
[image]

antwal, 2024-08-08:
@rikugun yes I saw that it was not available for arm64, I still tried to remove odbc but the result is this:

[image]