frangoteam/FUXA #1267: [BUG]

martinjug, 2024-07-08:
Hi,

I have found a few bugs:

**Describe the bug**

1. **Card Opening Position:**
   - On a desktop PC, the card opens in the correct position.
   - On an Android phone, all cards open in the bottom left corner.

Android version: ColorOS 13.1
Browser: Chrome ,Brave, Firefox

[image]
[image]

2. **Dialog Mode**:
   - In the Screen settings menu, if Align is set to "Top," the dialog mode opens correctly.
   - In the Screen settings menu, if Align is set to "Middle," only the overlay shadow is displayed, and no screen is shown. You get stuck in dialog mode without option to close it.

[image]

Platform: Windows, Android
Browser: Chrome ,Brave, Firefox, Edge

**Environment**
 - FUXA version: Version 1.2.0-1670
 - Platform/OS: Docker

unocelli, 2024-07-25:
Hi, Thanks for the report, we will try to fix it as soon as possible

unocelli, 2025-05-04:
I’m going to close this as resolved. let me know if you have any issues.

closed 2025-05-04 by unocelli