frangoteam/FUXA #1268: [BUG] Missing library SiemensS7

martinjug, 2024-07-08:
**Describe the bug**
Missing library SiemensS7.

**Screenshots**
[image]

**Environment**
-    FUXA version: Version **1.2.0-1724**
- Platform/OS: **Docker**

mauroalexandre, 2024-07-08:
I opened a discussion for voting on this topic a few days ago. Please consider participating.

https://github.com/frangoteam/FUXA/discussions/1265

martinjug, 2024-07-09:
It was included by default in Docker. Library is included in version **frangoteam/fuxa:1.1.19** and works without issue.
I think something got broken in latest version.

MatthewReed303, 2024-07-09:
@martinjug it's not in the packages for the latest https://github.com/frangoteam/FUXA/blob/master/server/package.json maybe @unocelli will add it so it can be installed separately like nodepccc I personally use OPC-UA with Siemens anyway, have for both s7-1200 and s7-1500

martinjug, 2024-07-11:
@MatthewReed303 The library worked perfectly, so I don't understand why it was excluded.
I'm currently using it to connect to Siemens LOGO, and there are still many machines/applications using S7 PLCs that don't have built-in OPCUA Server.

MatthewReed303, 2024-07-11:
@martinjug you could just fork Fuxa and add it back to the package list and do a build of your fork, remember to change the git clone in the docker file to your fork. If it works with no issues then create a pull request to add it back in or any fixes etc

unocelli, 2024-07-12:
Hi, the package have compatibility issue with the nodejs version 16/18, I propose to create a docker version (tagged s7) with node-snap7

martinjug, 2024-07-12:
@unocelli that would be great if there is S7 docker version based on node version 14.

unocelli, 2024-07-25:
Hi, you will find the the docker version with installed node-snap7 library in frangoteam/fuxa:snap7

martinjug, 2024-07-29:
Thanks @unocelli.
I have tested it and it works.

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.

closed 2024-08-23 by unocelli