frangoteam/FUXA #1282: New Feature - Ability to access historical tags in script

adnanebrahimi, 2024-07-20:
**Describe the feature**
This is a missed feature for the script engine to access historical tags in the script. 
by accessing historical tags in the script, you can now calculate historical data and show them in the view.

**Describe the solution you'd like**

Creating a new system function` $getHistoricalTags` that accepts three arguments:

1. Array of historical Tags
2. Date for past date
3. Date for recent date

**Usage:**
`console.log(await $getHistoricalTags(['t_7b1017a7-10924aca','t_5ed33907-40b642d3'],'2024/06/01 - 00:00:00','2024/06/30 - 23:59:59'));`

With this line of code, you will get historical data for June

Or you can get historical data until present time by passing **empty string** like this:

`console.log(await $getHistoricalTags(['t_7b1017a7-10924aca','t_5ed33907-40b642d3'],'2024/06/01 - 00:00:00',''));`

**Output**:

```json
{
  "t_7b1017a7-10924aca": [
    { "x": "2024-06-30T20:30:00.000Z", "y": "1"},
    { "x": "2024-07-09T13:08:45.822Z", "y": "0" }
   ],
    "t_5ed33907-40b642d3": [
    { "x": "2024-06-30T20:30:00.000Z", "y": "0" },
    { "x": "2024-07-09T13:08:45.822Z", "y": "0" }
  ]
}
```

MatthewReed303, 2024-07-20:
@adnanebrahimi why not use the odbc driver and connect to the DB and make the required SQL query? I'm sure Fuxa is using SQLite, i'm sure @unocelli can confirm this and how you could connect to the DB directly

adnanebrahimi, 2024-07-20:
Sometimes you don't have odbc database.

MatthewReed303, 2024-07-20:
@adnanebrahimi the odbc is now built into the latest master of Fuxa, if you do a docker build of the master you will have odbc and all the odbc drivers installed automatically. All you have todo is create a device, select odbc and enter the connection string. See https://github.com/frangoteam/FUXA/issues/1204 for more details and how to use. I will talk with @unocelli about adding an how to use to the wiki

closed 2024-07-26 by adnanebrahimi