frangoteam/FUXA #1285: [FEATURE] Bind-able property value in GaugeRangeProperty

henjoe, 2024-07-21:
**Describe the feature**
While creating a Faceplate view (a reusable view) on my project, one thing I encountered is the property of any element (for example a Gauge) cannot be binded to a "placeholder" or an specific "Tag". The property is somehow manually input, which works fine in some scenario. 
For example screenshot below, only the property "value" is bind-able. but the Min and Max is not.
[image]

I am creating a specific faceplate like this below, but the problem is every Tag has different minimum and maximum value, so I want the min and max value of my Bar gauge (as seen on my picture) to be dynamic also. 
[image]

**Describe the solution you'd like**
Add code to bind other property of instead of value only.

henjoe, 2024-07-21:
This will open a whole new level of SCADA for Fuxa 🔥 

Imagine if all property of any elements (charts, gauges, output, input and etc) can be binded to any tags in Fuxa.

unocelli, 2024-08-01:
Hi, a similar request already exists #609

henjoe, 2024-08-01:
> Hi, a similar request already exists #609

Its different, he is just pertaining to a progress bar change color based on min/max value of a single tag. 

While what I am suggesting is all properties of GaugeRange can be bind to a placeholder/tags instead of Tag value only. 

For xample, the min/max property can be bind to @max or @min placeholder.

jay-jjwu, 2024-08-20:
Intead of binding guage property to a specific tag, maybe another alternative is to bind guage property to scripts. This will lead more dynamic option to user for engineering. With scipts, write & read property, you can create any controls and shapes by user customized needs.

unocelli, 2024-09-03:
Hi, we are working to widget, where you can import svg image (widget) which has its own logic (javascript) and can be configured via parameters

henjoe, 2024-09-03:
> Hi, we are working to widget, where you can import svg image (widget) which has its own logic (javascript) and can be configured via parameters

That's a really nice idea! 
Can't wait to see it.