frangoteam/FUXA #1286: [Question] How to add custom font style?

henjoe, 2024-07-22:
I noticed that the font in the SVG editor is limited, I wonder if we can add more? like downloading the .tff and woff file putting it on a specific folder then it will be made available in the font-family options:

[image]

I tried to add a 7 segment font style under the
` src/assets/fonts/`
and try to npm start to the client folder, however I don't see the newly added fonts on the list. Maybe I just miss something here, can you guide us how to add custom font? 

Thank you.

Regards,

LordNetro, 2024-07-22:
Hi @henjoe , 

You must add the font to `client\src\styles.css` and in the `client\src\app\_helpers\define.ts`:

Example adding **Saira-Bold**:
- styles.css
```css
@font-face {
    font-family: "Saira-Bold";
    src: local("Saira-Bold"),
        url("./assets/fonts/Saira/static/Saira-Bold.ttf")
            format("truetype");
}
```

- define.ts
```ts
static fonts = ['Sans-serif', 'Roboto-Thin', 'Roboto-Light', 'Roboto-Regular', 'Roboto-Medium', 'Roboto-Bold', 'Quicksand-Regular', 'Quicksand-Medium', 'Quicksand-Bold', 'Andria-Signature', 'Saira-Bold'];
```
[image]

This is working for me, I hope it helps you :).

henjoe, 2024-07-22:
Hi Thanks For this! highly appreciate it! 🥳🥳  I will try this!

henjoe, 2024-07-22:
Hi @LordNetro I don't know if I am missing something, I can see now the newly added font style in the SVG editor, however it feels like it doesn't apply the font style? 

Below I can see the 7 segment option:
[image]

- Here is my code on `style.css`

```
@font-face {
    font-family: "7-segment";
    src: local("7-segment"), local("7-segment"),
        url("./assets/fonts/7segment/7segment.woff")
            format("woff"),
        url("./assets/fonts/7segment/7segment.ttf")
            format("truetype");
}
```

-  `define.ts`
`static fonts = ['Sans-serif', 'Roboto-Thin', 'Roboto-Light', 'Roboto-Regular', 'Roboto-Medium', 'Roboto-Bold', 'Quicksand-Regular', 'Quicksand-Medium', 'Quicksand-Bold', '7-segment'];`

I tried to apply the custom font but seems like not working:
[image]

Am I missing something here? or my font style is the problem?

LordNetro, 2024-07-22:
Hi @henjoe , i don't really know what are you missing, maybe is a Font problem, when I tried in the past with other fonts some didn't work, maybe you are using the variable font instead of the static ones? Or maybe the font doesen't support the font size used. Could you please try with another font please?

P.D: I'm using full stack mode on Fuxa (npm start on client folder and use the debug "Server & Client" option in VS Code.

henjoe, 2024-07-22:
> Hi @henjoe , i don't really know what are you missing, maybe is a Font problem, when I tried in the past with other fonts some didn't work, maybe you are using the variable font instead of the [...]

Okay I wil try and test on other font style. 

Thanks again brother.

henjoe, 2024-07-22:
Hi @LordNetro  I think the font-style has issue, I tried other and it works fine! 
Thanks for your help! I really appreciate it!

closed 2024-07-22 by henjoe

jingshui127, 2025-07-27:
> 嗨，我不知道我是否错过了什么，我现在可以看到 SVG 编辑器中新添加的字体样式，但感觉它不适用字体样式？
> 
> 下面我可以看到7段选项： [image]
> 
> * 下面是我在 style.css 上的代码
> 
> ```
> @font-face {
>     font-family: "7-segment";
>     src: local("7-segment"), local("7-segment"),
>         url("./assets/fonts/7segment/7segment.woff")
>             format("woff"),
>         url("./assets/fonts/7segment/7segment.ttf")
>             format("truetype");
> }
> ```
> 
> * `define.ts`
>   `static fonts = ['Sans-serif', 'Roboto-Thin', 'Roboto-Light', 'Roboto-Regular', 'Roboto-Medium', 'Roboto-Bold', 'Quicksand-Regular', 'Quicksand-Medium', 'Quicksand-Bold', '7-segment'];`
> 
> 我尝试使用自定义字体，但似乎不起作用： [image]
> 
> 我是不是漏掉了什么？还是我的字体有问题

我也尝试过，的确没有起作用。