frangoteam/FUXA #1290: [BUG]

sgocken, 2024-07-23:
**Describe the bug**
Using a Proc. Eng. Object to just be a static object doesn't use Fill unless a point is attached to the object

**To Reproduce**
Steps to reproduce the behavior:
1. Add Proc Eng Object
2. Set Fill
3. Run project and object is still white
4. Attach point to object
5. Run and see the object now has color

**Expected behavior**
That fill on the properties page would fill even without a point

**Environment**
 - FUXA version: 1.2.0-1772
 - Docker image: frangoteam/fuxa

unocelli, 2024-08-01:
Hi, I suppose it has since been solved, tested with  frangoteam/fuxa:latest (1.2.0-1784) it works as expected.

sgocken, 2024-08-01:
Still not working for me in FUXA V.1.2.0-1784

Created new object from Proc Eng and configed as shown in first pic, all other tabs are default. Only added a fill color.

[image]
[image]

sgocken, 2024-08-01:
I figured it out. There needs to be a value set between min/max. In case someone else runs into this issue.

[image]

unocelli, 2024-08-01:
if you do not intend to associate the colours with a tag logic, it is better to use the colours tool 
[image]

sgocken, 2024-08-01:
Mind Blown! I never even noticed that toolbar down at the bottom.
Thanks for the help

unocelli, 2024-08-23:
I’m closing this issue since there is nothing feasible

closed 2024-08-23 by unocelli