frangoteam/FUXA #1291: Fuxa with Node Js unofficial x86 version

nex2000, 2024-07-23:
Hello everyone

I have several ARM banana boards and other 32bit mini PCs. I would like to know if by installing an unofficial 32bit nodejs version I can still make fuxa work.

Has anyone ever tried it? Is there a way I can do this?

Thank you

Paul

MatthewReed303, 2024-07-23:
@nex2000 use docker and you should be good to go, there are Docker 32bit builds for ARM

nex2000, 2024-07-23:
Thanks a lot Matthew 

I'll follow your advice

Regards
Paul

nex2000, 2024-07-25:
I managed to get fuxa working on an old 32 bit thinclient, I used an unofficial 32 bit version of nodejs and modified the docker file. I attach the code of the file.

Regards

`FROM debian:bookworm-slim

RUN apt-get update && apt-get install -y dos2unix curl xz-utils build-essential unixodbc unixodbc-dev

# Change working directory
WORKDIR /usr/src/app

# Download and extract Node.js v15.14.0
RUN curl -O https://unofficial-builds.nodejs.org/download/release/v16.13.0/node-v16.13.0-linux-x86.tar.xz && \
    tar -xJf node-v16.13.0-linux-x86.tar.xz && \
    mv node-v16.13.0-linux-x86 /usr/local/node-v16.13.0 && \
    ln -s /usr/local/node-v16.13.0/bin/node /usr/bin/node && \
    ln -s /usr/local/node-v16.13.0/bin/npm /usr/bin/npm && \
    ln -s /usr/local/node-v16.13.0/bin/npx /usr/bin/npx

RUN apt-get install -y git

# Clone FUXA repository
RUN git clone https://github.com/frangoteam/FUXA.git

# Convert the script to Unix format and make it executable
RUN dos2unix FUXA/odbc/install_odbc_drivers.sh && chmod +x FUXA/odbc/install_odbc_drivers.sh

WORKDIR /usr/src/app/FUXA/odbc
RUN ./install_odbc_drivers.sh

# Change working directory
WORKDIR /usr/src/app

# Copy odbcinst.ini to /etc
RUN cp FUXA/odbc/odbcinst.ini /etc/odbcinst.ini

# Clone node-odbc repository
RUN git clone https://github.com/markdirish/node-odbc.git

# Change working directory to node-odbc
WORKDIR /usr/src/app/node-odbc

RUN apt-get update && apt-get install -y python3 \
    python3-pip \
    python3-venv

# Install compatible versions of global npm packages
RUN npm install -g node-gyp && \
    npm install -g npm@8 && \
    npm install -g node-addon-api && \
    npm install -g @mapbox/node-pre-gyp

# Install dependencies and build node-odbc
RUN npm ci --production && \
    ./node_modules/.bin/node-pre-gyp rebuild --production && \
    ./node_modules/.bin/node-pre-gyp package

# Install Fuxa server
WORKDIR /usr/src/app/FUXA/server
RUN npm install

# Workaround for sqlite3 https://stackoverflow.com/questions/71894884/sqlite3-err-dlopen-failed-version-glibc-2-29-not-found
RUN apt-get update && apt-get install -y sqlite3 libsqlite3-dev && \
    apt-get autoremove -yqq --purge && \
    apt-get clean && \
    rm -rf /var/lib/apt/lists/* && \
    npm install --build-from-source --sqlite=/usr/bin sqlite3

# Add project files
ADD . /usr/src/app/FUXA

# Set working directory
WORKDIR /usr/src/app/FUXA/server

# Expose port
EXPOSE 1881

# Start the server
CMD [ "npm", "start" ]
`