frangoteam/FUXA #1292: The Fuxa doesn't autostart

Brandon-sistein, 2024-07-23:
When I reestart the PC doesn't start the aplication automatically.

I have FUXA installed at a raspberry pi with Docker and I used this command to autostart

`docker run -d --restart unless-stopped frangoteam/fuxa`

when I execute this comand

`sudo docker inspect -f '{{.HostConfig.RestartPolicy.Name}}' fuxa`

Show me this:

`always`

MatthewReed303, 2024-07-23:
@Brandon-sistein Docker will start Fuxa automatically without doing anything? no need to run `docker run -d --restart unless-stopped frangoteam/fuxa` on startup. I prefer to use Docker Compose to setup Fuxa and the volumes/network etc

Brandon-sistein, 2024-09-12:
Hello, I solved with this command:

`sudo nano /etc/rc.local`

I added this commands:

`#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Your command here
docker run -d -p 1881:1881 frangoteam/fuxa:latest

exit 0`

But now when I restart the operating system the project is deleted and I ahve to reload the JSON of the proyect

Brandon-sistein, 2024-09-23:
How can I make it? @unocelli

unocelli, 2024-09-23:
docker run -d -p 1881:1881 -v fuxa_appdata:/usr/src/app/FUXA/server/_appdata -v fuxa_db:/usr/src/app/FUXA/server/_db -v fuxa_logs:/usr/src/app/FUXA/server/_logs -v fuxa_images:/usr/src/app/FUXA/server/_images frangoteam/fuxa:latest

so you define witch folder is out of docker

closed 2024-10-17 by Brandon-sistein