frangoteam/FUXA #1293: [QUESTION] OpenCard multiple instance of View

henjoe, 2024-07-23:
I created a faceplate view with the following place holders:
1. @TAGNAME
2. @PV
3. @MV
4. @SV 

[image]

On my Main view (shape component), I added some "click" actions with "open card" and bind to the place holder:
[image]

I did the same on other component on my Main View but this time difference binding of Tags.

What happen is when I open card my Faceplate on the first component, it actually works fine and I have no issue.
[image]

But when I open the 2nd face plate, what happen is it overrides the first Faceplate that is already open. It seems like it is not a new instance of my Faceplate.

[image]

As you can see, 2 different binded tags, but they are now identical. 
Is it some sort of bugs or a normal behaviour?

Regards,

robsori, 2024-07-24:
Hi. It's an older problem. See here: https://github.com/frangoteam/FUXA/issues/744

henjoe, 2024-07-24:
I did it!!! Almost 2 days of debugging and coding. 
I will test it more first then will do the PR.

[image]

unocelli, 2024-08-01:
#308
#805

henjoe, 2025-04-10:
@unocelli I already have working program for this. 
The concept is to add contructor on the view, and replace the view id of the same view so that it doesn't conflict and it can open multiple instance of Card (Popup) view. 

The main refactor is in the View constructor, and in fuxa component openCard.  I can share you the code and simply resolve conflict, sorry not really familiar with the PR thing,  (the last time I check, the latest fuxa is totally different from what I have now), but if you have any questions I can answer it.

unocelli, 2025-04-10:
@henjoe It would be great if you could share with me how you solved it, when I analysed it it seemed quite challenging to solve it.

henjoe, 2025-04-12:
Hi @unocelli  please check PR #1687