frangoteam/FUXA #1299: [BUG] ModBus RTU InputRegister Addressing misfire

maxopa, 2024-07-25:
**Describe the bug**
Greetings!
I have a ModBus RTU input registers addressing error. I added 1 input register with address 201, it works fine. After adding a second input register with address 211 everything went wrong. An error popped up in the FUXA console about incorrect addressing in the modbus device.
After debugging my modbus server, I found that FUXA was trying to access address 203.

One input register works fine with any address, but two does not.

So, I had 2 inputs with addresses 201 and 211, and neither one showing a value, the modbus server was getting a request for address 203.
When I changed the addresses to 200 and 211 - the device got address 199.
Replacing 211 with another address had no effect.

By the way, there is nothing wrong with coil addressing.

**To Reproduce**
1. Add ModBusRTU Device
2. Add tag to InputRegister 201
3. Add tag to Input Register 211
4. Get nothing))

**Expected behavior**
FUXA should receive two values from modbus rtu server

**Environment**
 - FUXA version: 1.1.19-1619
 - Node.js version: v18.20.4
 - npm version: 10.7.0
 - Platform/OS: Windows 10 19045.2965
 - Browser: Chrome and Edge

henjoe, 2024-07-26:
I also encountered this error. I added HR1 to HR10, which works fine, when I added HR11, then the error comes, worse is it affects all other HR (1-10) has reading. 

Seems like modbus TCP is more likely stable than the RTU.

maxopa, 2024-08-19:
Dear developer, pls pay attention to this bug, it is critical for building system on your scada software.

henjoe, 2024-08-19:
To be honest, it is not recommended to directly connect your serial data to your computer (with SCADA), better put in between a gateway, might be a converter from rs485 to opcua, that way you will never had a problem.

maxopa, 2024-08-19:
> To be honest, it is not recommended to directly connect your serial data to your computer (with SCADA), better put in between a gateway, might be a converter from rs485 to opcua, that way you will never had a problem.

To be honest I already wrote fine modbus to opcua converter, but the system become more laggy, I don't like that, I need faster reaction. Also, I don't understand difference what software will listen serial port, scada or converter.

Anyway, this thread is about real bug in functioning of the base protocol for SCADA.

maxopa, 2024-09-26:
Dear developer, pls check this issu, it is critical.