frangoteam/FUXA #1302: Security [POSSIBLE ISSUE]

bogdanvaduva, 2024-07-29:
**Describe the bug**
I've seen that FUXA has security issues when it comes getting the data:
For example: http://ip_of_fuxa/socket.io/?token=null&EIO=3&transport=polling&t=P3zcTwJ.0&sid=2PzusTMQs3Hh9PaCAABZ
The result could contain private data which shouldn't be available.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to your browser, open fuxa link.
2. Activate developer tools
3. Go to Network page (developer tools)
4. See the data corresponding to the following link (could be different in your case) 
http://ip_of_fuxa/socket.io/?token=null&EIO=3&transport=polling&t=P3zcTwJ.0&sid=2PzusTMQs3Hh9PaCAABZ

**Expected behavior**
The resulting data should not contain any private data (IPs, usernames, passwords or anything else). 

**Environment**
 - FUXA version: probably all versions

unocelli, 2024-08-01:
Hi, this is not a issue but it is about configuring for your goal. you can install nginx proxy and enable https.

bogdanvaduva, 2024-08-01:
Yes it's true, but i was thinking about basic encoding of data so it's not sent in plain. My issue was just to make aware anyone who wants to use FUXA over Internet. Again great job!

unocelli, 2024-08-23:
I’m closing this issue since there is nothing feasible

closed 2024-08-23 by unocelli