frangoteam/FUXA #1306: [BUG] Scripting tag system functions not working properly

LordNetro, 2024-07-31:
**Describe the bug**
When you export a **device**, edit it manually and import it back, the new **tags** added are not appearing in the `getTag` and `setTag` system functions on the `Scripting` tab, but they appear if you select the **tag** in an element property from a **view**.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `Devices` tab, create an **internal device** and add some tags
2. Click on `Export as JSON`
3. Edit the JSON adding new tags on the new device
4. `Import` back the edited JSON
5. Go to `Scripting` tab, create a `new script` and go to `system functions`
6. Click on `setTag` or `getTag` and the new **tags** and **device** will not appear.

**Expected behavior**
Acces the manually added tags of the device.

**Screenshots**
Device
[image]

Script
[image]

View
[image]

**Environment**
 - FUXA version: 1.2.0-1751
 - Node.js version: 18.20.3
 - npm version: 10.7.0
 - Platform/OS: Server=Windows 10 Pro / Client=Windows 11 Pro
 - Browser: Microsoft Edge

LordNetro, 2024-07-31:
After some testing I found that it occurs with all the internal devices you create, regardless if you export it and edit it.

developrepogithub, 2024-08-08:
@unocelli Is there any way of creating like an Internal Device but running in the server? (without using the main Fuxa Server device)

LordNetro, 2024-08-08:
@unocelli Is there any way of creating like an Internal Device but running in the server? (without using the main Fuxa Server device)

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli