frangoteam/FUXA #1308: [BUG] Cannot remote control device using FUXA

zhiwie, 2024-08-01:
**Describe the bug**
After connecting all the registers from device to FUXA, the current timestamp shows successful connection for each register.

However, I failed to modify the values/status of the connected device from FUXA. 

There are two types of error for registers (PowerMachine and FanSpeed etc.)
**1. PowerMachine**
Error message showed in console:
[image]

To reproduce the error: 
https://github.com/user-attachments/assets/2e56ef45-2e1b-4a09-9466-712f961bbe30

**2. FanSpeed**
Error message showed in console:
[image]

To reproduce the error:
https://github.com/user-attachments/assets/a383338a-4f1b-4c59-9291-41ed7628a830

**Expected behavior**
The output should show the modified input without any error

**Environment**
 - FUXA version: Version 1.1.19-1619
 - Node.js version: v14.20.0
 - npm version: 6.14.17
 - Platform/OS: Windows
 - Browser: GoogleChrome

unocelli, 2024-08-01:
Hi, I can't confirm, for me it works as expected (tested with modrssim2 simulator)

zhiwie, 2024-08-14:
@unocelli Hello, could you please show me how did you bind the input control to device tag for ModBus RTU using the simulator? 

I might have missed some steps, but I'm not sure

unocelli, 2024-08-23:
https://youtu.be/a74vQYnI6g4?si=yvXXDZYmbjRIXrjD
https://youtu.be/HwKQXagmD_Y?si=mSl1Lsd7DXDnZCQ9

closed 2024-09-04 by zhiwie