frangoteam/FUXA #1309: [BUG]

andy8601, 2024-08-02:
**Describe the bug**
A clear and concise description of what the bug is.

logs The time does not match the system time
**Environment**
 - FUXA version: 1.2.0-1784
 - Node.js version: 
 - npm version:
 - Platform/OS:docker
 - Browser:chrome

andy8601, 2024-08-05:
System time The log time is UTC+0, which is inconsistent with the time zone of the server or the system time in the upper right corner.

andy8601, 2024-08-05:
The system log time is UTC+0, which is inconsistent with the time zone of the server or the system time in the upper right corner.

LordNetro, 2024-08-09:
> 1. 2.When the script runs periodically, the FUXA service crashes after saving the project。
>    **Environment**

Do you have any useful information displayed in the console?

unocelli, 2024-10-27:
I’m closing this issue since there is nothing feasible and due to inactivity.

closed 2024-10-27 by unocelli