frangoteam/FUXA #1311: [FEATURE] About window size self-adaption

qaiu, 2024-08-07:
**Describe the feature**
window size self-adaption 

**Describe the solution you'd like**
It can adapt to small screens or low resolution devices (mobile devices) instead of displaying scrollbars and only showing partial images.

The current adaptation requires triggering when the window size changes

**Additional context**
[image]

thanks!

henjoe, 2024-08-07:
There is a settings in the Layout related to Zoom. 
Just set it to "Automatic"

MatthewReed303, 2024-08-07:
@qaiu try to use multi view and adjust grid settings to suit your view. The multi view is setup to auto fit screens and shift grids automatically based on screen size

qaiu, 2024-08-08:
> There is a settings in the Layout related to Zoom. 
> Just set it to "Automatic"

Already set, but still not working

qaiu, 2024-08-08:
> @qaiu try to use multi view and adjust grid settings to suit your view. The multi view is setup to auto fit screens and shift grids automatically based on screen size 

It works, but it is not adaptive. However, it can be adapted to different resolutions through multiple

unocelli, 2024-08-17:
Hi, try the last version and explains again what is not working

qaiu, 2024-08-20:
Good! The new version is OK.
At version 1.9 even if autoresize is enabled, it doesn't work the first time you open it
> Hi, try the last version and explains again what is not working

closed 2024-08-20 by qaiu