frangoteam/FUXA #1313: [QUESTION] How exactly the Authetication via Token works?

henjoe, 2024-08-07:
I don't know how this authentication token really actually works, 

When it was disabled, I have no problem, the Authorization on any object will be forbidden, I understand this. 

But when enabled, let say for 15mins, it automatically Signout any users when it is inactive, which is in my point of view, there are screens that actually doesn't require to be operated every minute, there are screens that is there for monitoring or view only. So it is actually hassle for the operator to login every specific time just being inactive.

I tried the options "Only for editor" , as far as I understand it will only signout you if you are in the editor page not in the Home page. But still, even I am in the home page, it automatically signout me for being inactive. 

Now back to my question, how this actually really works? Am I correct in my understanding that it will automatically sign-out you when you are being inactive? 

[image]

unocelli, 2024-10-27:
Hi, is correct. Currently for your use case we do not have a mode to show values for authorizations activated but nobody is signed in. To do!

unocelli, 2025-05-04:
The feature is now available in versione 1.2.5.
'guest' authentication to show connected View in logout mode

closed 2025-05-04 by unocelli