frangoteam/FUXA #1314: [FEATURE] Write on lost "input value" element focus

martinjug, 2024-08-09:
**Write on Lost "Input Value" Element Focus**
Behavior of the element when entering a value:
- **Active**: The value is immediately written when the control element loses focus. Pressing the Esc key cancels the input, and the value is not written.
- **Inactive**: The value is only written after confirming with the Enter key. Pressing the Esc key cancels the input, and the value is not set.

**Describe the feature**
This feature enhances usability by reducing the chance of mistakenly thinking a value has been entered into a field. However, if you don’t press Enter afterward, the value isn’t actually applied, leading to a mismatch between the input field and the variable it should update.

[image]

**Additional context**
Problem
The value isn’t actually applied, leading to a mismatch between the input field and the variable it should update
[image]

unocelli, 2024-08-09:
Hi, if you take the current master branch and you set the flag 'Enable update' by ESC the input field will be update with the current value. However, I realise that this could be a problem for the case where you want to create a settings dialog with several input fields and a confirmation button to write values on the tags. 
So your proposal is the best solution. Thanks

wagod835512, 2024-08-17:
HI, @unocelli

I've tried the latest version 1.2.0, but it seems that it's still not working as we discussed. The option to write a value without pressing enter (Write on lost focus) is still not available.

unocelli, 2024-08-17:
Hi, sorry for the misunderstanding, for me you always have to confirm the value with Enter, it does not seem to me to be a feature applicable to an hmi or scada.

unocelli, 2024-10-27:
Let's close here, since there is nothing feasible

closed 2024-10-27 by unocelli