frangoteam/FUXA #1315: [BUG]I passed a value to a page and the click changed itself.

hxjackcn, 2024-08-10:
Earlier versions did not have this problem.

https://github.com/user-attachments/assets/42b7cb72-c13c-491b-88fe-b45be8f3f5f7

hxjackcn, 2024-08-10:
[image]

hxjackcn, 2024-08-14:
In this case, value transfer control cannot be achieved.

hxjackcn, 2024-08-15:
@unocelli

unocelli, 2024-08-17:
Hi, in thwe new versions we will use the placeholder, it is definitely an improvement. we will investigate about the support of old functionality with internal device

hxjackcn, 2024-08-18:
@unocelli I didn't notice the problem with placeholders before, it's great, I will use placeholders from now on.

closed 2024-08-18 by hxjackcn

unocelli, 2024-08-23:
I’m going to close this as resolved. let me know if you have any issues.